tpaviot/pythonocc-core #955: issues creating prisms in a circular pattern

jkbgbr, 2021-01-14:
I'm generating thin plates around a cylinder, with the plates being approximately radially aligned.
I get confusing results, as if during the steps some direction were reversed. The code to duplicate the issue is attached below. I run the latest version 7.4.1-dev.

The plates are rendered the following way:
- a set of points describing a rectangle is defined (5 points, first equals the last), used to generate all plates.
- for each plate to be rendered a starting point and a vector is given. The starting points lie on a circle, the vectors are the normals for the future plates. These are externally calculated and seem to be OK to me.
- a wire is created in the plane given by a starting point and the normal vector. 
- a face is created from the wire and a solid is made by sweeping the face along a vector.

The expected result would be a number of plates in a circular pattern, all in the same height position and with the same alignment. However, I get the plates strangely mixed up: rotated about the normal vector by 90, 180 degrees... also depending on the circumferential locations defined to generate the positions in the circular pattern; as if the cause was some kind of a numeric issue.

I'd welcome any insight for the possble cause of the results and how to avoid this.

Cheers,
Gábor

```

from OCC.Core.GCE2d import GCE2d_MakeSegment
from OCC.Core.GeomAPI import geomapi_To3d
from OCC.Core.gp import gp_Pnt, gp_Pln, gp_Dir, gp_Vec, gp_Pnt2d
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace, BRepBuilderAPI_MakeWire
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakePrism

# plate thickness
thickness = 10

# the points defining the rectangle
points = [
 (0.0, 0.0),
 (0.0, 300.0),
 (-200.0, 300.0),
 (-200.0, 0.0),
 (0.0, 0.0)
]

# plate definitions: circumferential position: {normal vector, starting point}
# use either of the values, results are different
# 0 - 30 - 60   -> really erratic results
posinfo = {0: {'normalvector': (0.0, -1.0, 0.0), 'origin': (761.0, 0.01, 1550)},
 30: {'normalvector': (0.5, -0.866025404, 0.0),
      'origin': (659.04033228, 380.508660254, 1550)},
 60: {'normalvector': (0.866025404, -0.5, 0.0),
      'origin': (380.491339746, 659.05033228, 1550)},
 90: {'normalvector': (1.0, 0.0, 0.0), 'origin': (-0.01, 761.0, 1550)},
 120: {'normalvector': (0.866025404, 0.5, -0.0),
       'origin': (-380.508660254, 659.04033228, 1550)},
 150: {'normalvector': (0.5, 0.866025404, -0.0),
       'origin': (-659.05033228, 380.491339746, 1550)},
 180: {'normalvector': (0.0, 1.0, -0.0),
       'origin': (-761.0, -0.01, 1550)},
 210: {'normalvector': (-0.5, 0.866025404, 0.0),
       'origin': (-659.04033228, -380.508660254, 1550)},
 240: {'normalvector': (-0.866025404, 0.5, 0.0),
       'origin': (-380.491339746, -659.05033228, 1550)},
 270: {'normalvector': (-1.0, 0.0, 0.0),
       'origin': (0.01, -761.0, 1550)},
 300: {'normalvector': (-0.866025404, -0.5, 0.0),
       'origin': (380.508660254, -659.04033228, 1550)},
 330: {'normalvector': (-0.5, -0.866025404, 0.0),
       'origin': (659.05033228, -380.491339746, 1550)}}

# # 0.01 - 30.01 - 60.01   -> less funny results
# posinfo = {0.01: {'normalvector': (0.000174533, -0.999999985, 0.0),
#         'origin': (760.999986664, 0.142819555, 1550)},
#  30.01: {'normalvector': (0.500151142, -0.865938124, 0.0),
#          'origin': (658.973910953, 380.623678695, 1550)},
#  60.01: {'normalvector': (0.866112657, -0.499848842, 0.0),
#          'origin': (380.376307969, 659.116730508, 1550)},
#  90.01: {'normalvector': (0.999999985, 0.000174533, -0.0),
#          'origin': (-0.142819555, 760.999986664, 1550)},
#  120.01: {'normalvector': (0.865938124, 0.500151142, -0.0),
#           'origin': (-380.623678695, 658.973910953, 1550)},
#  150.01: {'normalvector': (0.499848842, 0.866112657, -0.0),
#           'origin': (-659.116730508, 380.376307969, 1550)},
#  180.01: {'normalvector': (-0.000174533, 0.999999985, 0.0),
#           'origin': (-760.999986664, -0.142819555, 1550)},
#  210.01: {'normalvector': (-0.500151142, 0.865938124, 0.0),
#           'origin': (-658.973910953, -380.623678695, 1550)},
#  240.01: {'normalvector': (-0.866112657, 0.499848842, 0.0),
#           'origin': (-380.376307969, -659.116730508, 1550)},
#  270.01: {'normalvector': (-0.999999985, -0.000174533, 0.0),
#           'origin': (0.142819555, -760.999986664, 1550)},
#  300.01: {'normalvector': (-0.865938124, -0.500151142, 0.0),
#           'origin': (380.623678695, -658.973910953, 1550)},
#  330.01: {'normalvector': (-0.499848842, -0.866112657, 0.0),
#           'origin': (659.116730508, -380.376307969, 1550)}}

def create_wire(origin, normalvector):
    """
    Creates a wire in the plane defined by origin and normalvector
    """

    # Making the plane.
    plane = gp_Pln(gp_Pnt(*origin), gp_Dir(*normalvector))
    # making the segments
    segs = [GCE2d_MakeSegment(gp_Pnt2d(*p1), gp_Pnt2d(*p2)).Value() for p1, p2 in zip(points, points[1:])]
    # building the wire
    w = BRepBuilderAPI_MakeWire()
    for seg in segs:
        w.Add(BRepBuilderAPI_MakeEdge(geomapi_To3d(seg, plane)).Edge())
    wire_ = w.Wire()
    return wire_

def create_prism(wire_, pos_in, pos_out):
    """
    Creates a prism between pos_in and pos_out, from the given wire.
    creator_wire is a Wire that is turned to a face first
    """

    # face from wire
    _shape_in_progress = BRepBuilderAPI_MakeFace(wire_)
    face_to_sweep = _shape_in_progress.Face()
    # creating the plate
    _shape_in_progress = BRepPrimAPI_MakePrism(face_to_sweep,
                                        gp_Vec(gp_Pnt(*pos_in),
                                               gp_Pnt(*pos_out))).Shape()
    return _shape_in_progress

if __name__ == '__main__':

    _shapes = []
    _wires = []
    for circpos, data in posinfo.items():
        wire = create_wire(data['origin'], data['normalvector'])
        _wires.append(wire)
        _pin = data['origin']
        meridian = tuple([thickness * x for x in data['normalvector']])
        _pout = tuple([x + y for x, y in zip(_pin, meridian)])
        _shapes.append(create_prism(wire, pos_in=_pin, pos_out=_pout))

    from OCC.Display.SimpleGui import init_display
    display, start_display, add_menu, add_function_to_menu = init_display()
    display.DisplayShape(_wires, update=True)
    display.DisplayShape(_shapes, update=True)
    start_display()

```

Results obtained using the inputs 0 - 30 - 60...
The green circle shows the base of the pattern, all plates should be on the outside
[image]

same result, with the alignments better visible
[image]

OkanPinar, 2021-01-14:
I could not figure out why you are describing make prism operation as sweep. Do you want to create angular slice with variable thickness or simply extruded plate from mid plane with constant thickness?

jkbgbr, 2021-01-15:
I guess I call it sweep b/c its a sweep along a straight line.
The plates should be identical and the pattern pointsymmetric.

I found a workaround - a possibly much simpler solution: the plates are created "locally", that is, the plate solids are created at the origin, in a plane aligned with the X-Z axis and once this solid is created is there a transformation (translation and rotation in one step). This produces much better results, the parts are flipped only at multiples of 360 degree - and this can be dealt with by simply using angular values slightly off, e.g. 360 -> 360+1e-5.

closed 2021-01-15 by jkbgbr