tpaviot/pythonocc-core #956: How do you specify which Python version(s) to build against?

salfter, 2021-01-21:
The build system appears to just select the newest available Python, as far as I've been able to tell.  It's fairly common to have multiple versions installed, but to want most things built against an older, more stable version.  Gentoo Linux normally keeps things straight with its distutils-r1 eclass, but right now I have a host where pythonocc was built against Python 3.9 while cadquery (and most everything else) was built against Python 3.8.  How do I get pythonocc to build against a specific Python version (or, potentially, multiple versions)?

salfter, 2021-01-21:
Passing -DPython3_EXECUTABLE=/usr/bin/python3.8 to cmake looks like it might work to select one version.  How about multiple versions?

rainman110, 2021-01-22:
> Passing -DPython3_EXECUTABLE=/usr/bin/python3.8 to cmake looks like it might work to select one version. 

This is correct.

The project cannot be built for multiple versions at the same time. This is technically impossible. What you can do is to create multiple build directories for the different python versions.

Dr-Terrible, 2021-01-22:
@salfter In my overlay you can find a `pythonocc` ebuild which properly handles multi-python installations, as required by Gentoo QA (python modules must be byte-compiled): https://github.com/Dr-Terrible/ineluctable-overlay/tree/master/dev-python/pythonocc

It's far from being perfect, but it's a starting point and compile and install `pythonocc` correctly as required by upstream. Feel free to use it and test it. Feedback are welcome.

You can also find ebuilds for CadQuery and CQ-editor, in case you want to test all the features of `pythonocc` extensively.