tpaviot/pythonocc-core #958: ImportError: DLL load failed

songyuke1990, 2021-01-30:
After install 7.4.1 by conda creating environment named pyocc, the code I run:
from OCC.Display.SimpleGui import init_display
in ..\Anaconda\envs\pyocc\python.exe 
raise:
    from . import _Aspect
ImportError: DLL load failed

I can't find the way to solve this problem, can you help me?
Thank You.

This is my pyocc's packages:
[image]

songyuke1990, 2021-01-30:
my conda list:
# Name                    Version                   Build  Channel
ca-certificates           2020.12.5            h5b45459_0    conda-forge
certifi                   2020.12.5        py37h03978a9_1    conda-forge
fontconfig                2.13.1            h1989441_1004    conda-forge
freeimageplus             3.18.0               h6538335_2    dlr-sc
freetype                  2.10.4               h546665d_1    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libxml2                   2.9.10               hf5bbc77_3    conda-forge
occt                      7.4.0                h823b557_3    conda-forge
opencascade               7.4.0                he980bc4_5    dlr-sc
openssl                   1.1.1i               h8ffe710_0    conda-forge
pip                       20.3.3           py37haa95532_0    defaults
pyqt5                     5.15.2                   pypi_0    pypi
pyqt5-sip                 12.8.1                   pypi_0    pypi
python                    3.7.9                h60c2a47_0    defaults
python_abi                3.7                     1_cp37m    conda-forge
pythonocc-core            7.4.1            py37hc019675_0    conda-forge
setuptools                52.0.0           py37haa95532_0    defaults
six                       1.15.0           py37haa95532_0    defaults
sqlite                    3.33.0               h2a8f88b_0    defaults
tbb                       2020.2               h2d74725_3    conda-forge
vc                        14.2                 h21ff451_1    defaults
vs2015_runtime            14.27.29016          h5e58377_2    defaults
wheel                     0.36.2             pyhd3eb1b0_0    defaults
wincertstore              0.2                      py37_0    defaults
zlib                      1.2.11               h62dcd97_4    defaults

songyuke1990, 2021-01-30:
This time, I create a test.py under the directory of _Aspect.pyd, and replace the code 
from . import _Aspect    >>>TO >>> from . import test
and the test.py can run.
So it isn't the PATH or PYTHONPATH error setting, maybe the code in _Aspect.pyd could not run.
Other modules have the same problem.

tpaviot, 2021-01-30:
The ```Aspect``` module is linked to OpenGl. Try the following:
```python
from OCC.Core.gp import gp_Pnt
```
If this works, then the problem comes from your Opengl driver. If it doenst work, this come from a path issue.

songyuke1990, 2021-01-30:
> The `Aspect` module is linked to OpenGl. Try the following:
> 
> ```python
> from OCC.Core.gp import gp_Pnt
> ```
> 
> If this works, then the problem comes from your Opengl driver. If it doenst work, this come from a path issue.

I import gp_Pnt still raise:
    from . import _gp  ImportError: DLL load failed:

As you said, it's a path issue, how to deal with it

songyuke1990, 2021-02-01:
> I import gp_Pnt still raise:
> from . import _gp ImportError: DLL load failed:
>
> As you said, it's a path issue, how to deal with it

I finally solve the path issue in this way:
In pycharm,   In  run>>config>>environment variables , ADD "PATH=...\Anaconda3\envs\pyocc\Library\bin"

but new error raise, and i will try to deal latter :<
RuntimeError: Aspect_GraphicDeviceDefinitionErrorOpenGl_Window::CreateWindow: SetPixelFormat failed. Error code: 2000 raised from method Init of class Display3d

songyuke1990, 2021-02-02:
The openGI problem has also been solved by this way:    :)  :)  :)
I replace 7.4.1 version to 7.4.0 version, then it works   :)  :)  :)

closed 2021-02-02 by songyuke1990

Subhashplcm, 2024-02-29:
@songyuke1990 ;
I am currently working on a academic project where i have to visualize the .step file. I am facing the same issue, Could you please let me know how to resolve this.
I have very little experience in programming, may i know how to visualize the .step file using pythonocc.