tpaviot/pythonocc-core #964: Compilation failing on OSX 10.15.6 (Catalina) with error "<rapidjson/prettywriter.h> not found"

rohan-kekatpure, 2021-02-16:
I'm new to CMake but familiar with compilation of C++ projects from source. When trying to compile pythonocc from source on OSX-catalina, the compilation fails with the following error

```
<rapidjson/prettywriter.h> not found
```

I installed rapidjson using `brew install rapidjson` and confirmed that the required files (specifically `prettywriter.h` was present in the include directories. Yet CMake was not including the header files in `/usr/local/include` (my guess).

Adding

```
include_directories("/usr/local/include") 
```

inside the 

```
if(APPLE)
endif(APPLE)
```

solved the issue.

tpaviot, 2021-02-16:
Thank you for the report. Which pythonocc-core version are you trying to build ?

rohan-kekatpure, 2021-02-16:
I followed the instructions in [Install.md](https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md) and cloned the latest repo. So (I guess) Version 7.4.1.

tpaviot, 2021-02-16:
No, the current master branch is the dev version, to be 7.5.0

rohan-kekatpure, 2021-02-16:
Sorry. Yes, I built the current master branch.

fslurrehman, 2021-02-21:
Hi, 
I was compiling latest source of pythonocc on debian buster and face this issue. I installed rapidjson-dev and applied make command again and issue was resolved. 

I think there is need of adding dependency of rapidjson in the installation instructions. Like for debian, before compilation, install the following package:
`sudo apt install rapidjson-dev`

FR

howardjones, 2021-07-09:
Just ran into the same thing with building 7.5.1 - @fslurrehman is correct for Ubuntu.
(I hope this library does what I want - compiling it and opencascade has been quite an adventure! :-) )

oleg-medovikov, 2022-01-12:
Hi, I ran into such a problem on a mac m1. I did 
`brew install rapidjson`
but it didn't help.
[image]

lotuc, 2023-11-10:
Manually configuring the [CPP_INCLUDE_PATH](https://stackoverflow.com/questions/558803/how-can-i-add-a-default-include-path-for-gcc-in-linux) works for me.

```
export CPLUS_INCLUDE_PATH=/usr/local/include:$CPLUS_INCLUDE_PATH
```

Andrej730, 2024-07-12:
On Windows I was able to solve the similar issue with
```powershell
$env:CL = "/I""X:\occt-7.8.0-vc143-64\3rdparty-vc14-64\rapidjson-1.1.0\include"" $env:CL"
msbuild /p:Configuration=Release .\PYTHONOCC.sln
```