tpaviot/pythonocc-core #965: Step file importer only imports shape but not path on shape

Thieso, 2021-02-17:
Hello, 

I would like to import a step file with a shape and a path on that shape. In FreeCAD that works as intented as shown here:
[image]

However when I import the file in pythonocc and show it, it does not show the path on the surface, see here:
[image]

I use this code to display the file 

```
import os.path
from IPython.core.display import SVG
from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer
from OCC.Extend.DataExchange import read_step_file

shp = read_step_file(os.path.join(<path to stp file>)
my_renderer = JupyterRenderer(size=(800,700))
my_renderer.DisplayShape(shp, render_edges=True)
my_renderer.Display()
```

Can you tell me why this does not work and how to fix it? 

Thanks a lot

JorisCodde, 2021-02-18:
Maybe it's a display issue and not a step importer issue ?

What's in the shp variable ?

Thieso, 2021-02-19:
It is a object of <class 'TopoDS_Compound'>, or what do you mean?

closed 2021-04-01 by Thieso