tpaviot/pythonocc-core #968: Updating installation requirments for opencascade and cmake version

fslurrehman, 2021-02-21:
Hi, 

Two updates are required. 
1. Version in install.md for opencascade
2. cmake version

1. There is need of updating installation requirement for opencascade version from 7.4.0 to 7.5.0. The compilation requires 7.5.0. It confused me as I downloaded and compiled 7.4 and then faced this issue in compilation of pythonocc. 

2. Also the cmake_policy(SET CMP0086 OLD) in CMakeList.txt is not recognized in cmake 3.12 and 3.13.4 (which I have in debian). Also there is error of some missing target in cmake building processing when cmake run code at line 261 for Model Toolkits. There are many errors regarding swig like:
` Cmake Error at /usr/share/cmake-3.13/Modules/Use SWIG.cmake:665 (add_library):Target "_ShapeProcess" links to target "Python3::Module" but the target wasnot found. Perhaps a find_package() call is missing for an IMPORTEDtarget, or an ALIAS target is missing`

These error of old cmake policy and target were resolved when I downloaded and used latest version of cmake.org.

I was compiling source from github.  

Regards,

FR

tpaviot, 2021-03-18:
Thank you for your feedback. Fixed by commits b815496c and 8475f38f.

I don't understand the ``_ShapeProcess`` issue. Is it just a warning or an error (build fails)?

closed 2021-03-18 by tpaviot

fslurrehman, 2021-03-18:
> Thank you for your feedback. Fixed by commits [b815496](https://github.com/tpaviot/pythonocc-core/commit/b815496c1395f8b7c243784ff60699e7151b6037) and [8475f38](https://github.com/tpaviot/pythonocc-core/commit/8475f38f4cab40dd1361942e6470512599740359).
> 
> I don't understand the `_ShapeProcess` issue. Is it just a warning or an error (build fails)?

Thanks for resolving it. 
`_ShapeProcess` issue was build error.

commit 8475f38f (2021-03-18): Update occt version number in INSTALL.md Fix #968