tpaviot/pythonocc-core #969: CentOS8 pythonocc-core compilation error

Rom1Linux, 2021-02-24:
Bonjour/Hello
I try to compil & install pythonOCC-core on CentOS 8, 
it's ok OCE-0.18.3

i use this command:

`cmake -DOCE_INCLUDE_PATH=/usr/local/include/oce -DOCE_LIB_PATH=/usr/local/lib/oce-0.18 -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib64/libpython3.6m.so -DPYTHONOCC_INSTALL_DIRECTORY:PATH=/usr/lib/python3.6/site-packages/OCC ..`

The result and error is:

```
Make Error at CMakeLists.txt:38 (cmake_policy):
  Policy "CMP0086" is not known to this version of CMake.

CMake Error at CMakeLists.txt:49 (cmake_policy):
  Policy "CMP0086" is not known to this version of CMake.

--  
-- Python3 interpreter:/usr/bin/python3.6
-- Python include directory: /usr/include/python3.6m
-- Python library release: /usr/lib64/libpython3.6m.so
-- Disabled SWIG warnings
-- Disabled deprecation warnings for oce
-- pythonocc modules will be installed to: /usr/lib/python3.6/site-packages/OCC
-- Configuring incomplete, errors occurred!
```
`
I have no logs to understand why, can you help me ?

Rom1Linux, 2021-02-24:
for informations:
cmake 3.13.5
Swig 3.0.12
Python 3.6.8

tpaviot, 2021-03-18:
Thank you for reporting this issue.

tpaviot, 2021-03-18:
Fixed by commit b815496c

commit b815496c (2021-03-18): Fix compilation for old versions of cmake. Fix #969
closed 2021-03-18 by tpaviot