tpaviot/pythonocc-core #972: how do i close SimpleGUI window?

RobinVanTol, 2021-03-04:
The win object in the init display function does not get returned and there is also no method made to close the window. Im using pyQt5 as backend which has the close function, so in init_display win.close() would close the window, however i dont see any way to do this with what init_display returns.

Is there a way to close the window?

tpaviot, 2021-03-18:
```python
sys.exit() # this will close the window, for sure
``` 
😃

RobinVanTol, 2021-03-26:
I mean, well yes, but i only want to close the simplegui window, not the entire program

tpaviot, 2021-04-14:
This is a PyQt5 development issue. Don't use the SimpleGui, but build your own gui using the qtViewer3d widget.

closed 2021-04-14 by tpaviot