tpaviot/pythonocc-core #975: Intersection Between Two Shapes

Vahid44, 2021-03-12:
Hello everyone,
Is there any way to find out if the intersection between two shapes is Null or not? (see the result as a boolean)
I searched "TopologyExplorer" and "BRepAlgoAPI_Section" but none of them has a specific function for this purpose.

Thanks,
Vahid

Vahid44, 2021-03-12:
Hello @OkanPinar,
Do you know any function for this issue?

tpaviot, 2021-03-18:
Have a look at the example https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_boolean.py

Vahid44, 2021-03-19:
Thank you very much dear @tpaviot.
Actually, I searched this page but couldn't find a specific function for my problem.

I know this can be done by using other functions. For example, using BRepAlgoAPI_Section and TopologyExplorer and see if the Section has any edges, points, etc.

But I was wondering if there is any particular function for solving this problem since I guess it would be very useful in many cases.

Regards,
Vahid

tpaviot, 2021-03-19:
IMO the most robust way to check for intersections between two shapes is to mesh both shapes to a triangular mesh and use a mesh library to perform boolean operations/minimum distance (e.g. trimesh).

Vahid44, 2021-03-19:
Thank you dear @tpaviot

I'm working with IFC objects and use pythonocc to define objects' shapes. 
Is that possible to use trimesh for meshing OCC shapes? 
Or I need to use other meshing libraries, like SMESH (I don't know if it is still supported by pythonocc or not.)

tpaviot, 2021-03-19:
You can use the builtin pythonocc mesher or a third one (I use to work with gmsh when accurate meshes are needed).

closed 2021-04-14 by tpaviot