tpaviot/pythonocc-core #976: Offscreen Renderer need "two passes" to render correctly

rhesus-ltd, 2021-03-13:
Hi there

I'm using:
```
pythonocc-core            7.4.1            py37h2700f40_0    conda-forge
```
on Ubuntu 18.04.2 LTS

I see a strage behaviour I'm not sure if this is a bug or a mismatch in my setup.

I have the following code in a Jupyter-Notebook:
(A:)
```
from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer
from OCC.Extend.DataExchange import read_step_file_with_names_colors
from OCC.Core.Graphic3d import Graphic3d_BufferType
from OCC.Display.OCCViewer import Viewer3d

from PIL import Image
from PIL import ImageOps

my_renderer = JupyterRenderer()

W = 196
H = 128

filename = 'tactile_switch_3.5x2.9_4F.stp'
shapes_labels_colors = read_step_file_with_names_colors(filename)
```
and use the following to render an image of the step file:
(B:)
``` 
offscreen_renderer = Viewer3d()
offscreen_renderer.Create()
offscreen_renderer.SetSize(W, H)
offscreen_renderer.SetModeShaded()

for shp in shapes_labels_colors:
    label, c = shapes_labels_colors[shp]
    offscreen_renderer.DisplayColoredShape(shp, color=c, update=True)  

data_raw = offscreen_renderer.GetImageData(W, H, Graphic3d_BufferType.Graphic3d_BT_RGB)
img = Image.frombytes('RGB', (W, H), data_raw)
img = img.rotate(180, expand=True)
```

Now if I look at the image it looks like this (regardless of jupyter-Renderer or written to disk using Pillow:
[image]

The magic happens when I run the Block B a second time. Then the image becomes:
[image]

This can be reproduced outside of jupyter as well. 

But: If I use View.Dump to write a PNG file the file is correct the first time.

Thanks and feel free to close if this is not the right place to mention such an issue.

tpaviot, 2021-03-18:
I'm not aware of this issue, thank you for providing the STEP file I'll have a look.

pridi, 2021-04-02:
@rhesus-ltd This is not completely the same issue as yours but on Windows when I dump the image everything is correct but when I use Pillow I obtain this result.

[image]

@tpaviot do you have news about this?