tpaviot/pythonocc-core #977: FitAll() strange result

pridi, 2021-03-13:
I'm having a very strange issue when apply a display.FitAll() to two simple shapes. The first shape is a cube (CUBE_ON_ZERO.stp) generated by an external CAD in the world position (0,0,0). When I try to display.FitAll() everything is correctly shown: the shape is properly fitted in the view.

[image]

The second cube (CUBE_ROTATED) has been generated with the same external CAD, the only difference is that it 
[Archive.zip](https://github.com/tpaviot/pythonocc-core/files/6134256/Archive.zip)

has been generated on a plane ROTATED on random angles. The strange thing is that when I apply the display.FitAll() the image is not completely fitted in the view:

[image]

Even if I try to orient the shape in the view and apply a FitAll, nothing changes: the shape is not fitted.
Could you please help me to solve the issue? (attached the two STEP files)

tpaviot, 2021-03-18:
Can you please explain what you expect from the second screenshot? It seems the shape actually fits the window size.

pridi, 2021-03-19:
If you look at the second image there's a lot of white space around the model, even if the FitAll has been applied to the display. 
I've just figured out that this effect is present for ALL models that have been generated with axes NOT orthogonal to the XYZ world axes. By applying a transformation to the model to have its axes orthogonal to the world axes, the FitAll works correctly.

tpaviot, 2021-03-19:
I think this is due to the bounding box computation used to fit the view. It must be an Aligned bounding box, maybe an oriented bounding box would be better. This is done at the c++ level (in the occt library), not pythonocc. You may have to write your own FitAll method using an Oriented Bounding Box

tpaviot, 2021-04-14:
You can also use a camera-specific viewpoint, see related examples at https://github.com/tpaviot/pythonocc-demos/tree/master/examples

closed 2021-04-14 by tpaviot

pridi, 2021-04-17:
Thank you @tpaviot. I've followed your idea by creating a camera-specific viewpoint.