tpaviot/pythonocc-core #979: shapes recognition

alexinteam, 2021-03-17:
Hi! I'm trying to identify objects in my model. How can I do this? I discovered examples and identifing holes is clear for me. But what should be done if Object I want to discover is severan plain. 
[image]
PS. I make my model from step file

tpaviot, 2021-03-18:
Have a look at the example https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_geometry_recognize_feature.py

alexinteam, 2021-03-19:
Hello!, Thanks for the advice. I did, but still I'm wondering if it possible to detect cube hole. Cube hole is a List of plane faces. Is there algoritm to concat those faces and recognize that it is cube hole?

tpaviot, 2021-03-19:
There's not an out-of-the-box algorithm, but all the bases are available to build what you need.

closed 2021-03-22 by alexinteam