tpaviot/pythonocc-core #980: off-screen rendering on Mac - feature request

pridi, 2021-03-19:
@tpaviot @jf--- Hello all, do you have any update on the possibility to have an offscreen rendering on Mac? Do you know, alternatively, a way to have a byte array containing the "screenshot" of the rendered part? #

tpaviot, 2021-04-04:
So far, the offscreen renderer only supports linux and windows. osx requires a specific fix.

tpaviot, 2021-04-04:
I'm not sure the off-screen rendering has the same meaning on windows and Linux than on osx. On windows/linux, you can have machines without a graphic installed, or no X server, which is not the case for apple devices. As a consequence, it is always possible to create a ``NSView``. What you need, IMO, is a "render to image" feature. Did you try something like:

```python
from OCC.Display.SimpleGui import init_display
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

display, start_display, add_menu, add_function_to_menu = init_display()
my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()

display.DisplayShape(my_box)#, update=True)
display.View.FitAll()
display.View.Dump("box.jpg")

```

pridi, 2021-04-17:
@tpaviot Thank you. By simply avoiding to show on the screen the window, you can make the dump and create an offscreen render. The thing 'm not able to do is to make the offscreen render without saving the image on disk, keeping the image data stored in a variable, like in your demos.