tpaviot/pythonocc-core #981: Uncostistent pickling for thicksolid result.

mirmik, 2021-03-20:
Hello. 

I found a case in which pickle does not correctly restore the serialized object. 
It seems that the resulting object is turned inside out.

`#!/usr/bin/env python3

import OCC.Core.BRepPrimAPI
import OCC.Core.BRepOffsetAPI
import OCC.Core.BRepBuilderAPI
import OCC.Core.TopTools
import OCC.Core.TopExp
import OCC.Core.gp
import OCC.Display.SimpleGui
from OCC.Core.TopAbs import TopAbs_FACE

import pickle

box = OCC.Core.BRepPrimAPI.BRepPrimAPI_MakeBox(10,10,10).Shape()
facesToRemove = OCC.Core.TopTools.TopTools_ListOfShape()

ex = OCC.Core.TopExp.TopExp_Explorer(box, TopAbs_FACE)
face = ex.Current()

facesToRemove.Append(face)

algo = OCC.Core.BRepOffsetAPI.BRepOffsetAPI_MakeThickSolid()
algo.MakeThickSolidByJoin(box, facesToRemove, 1, 1.e-3)

thick_box = algo.Shape()

s = pickle.dumps(thick_box)
thick_box_2 = pickle.loads(s)

display, start_display, add_menu, add_function_to_menu = OCC.Display.SimpleGui.init_display()

trsf = OCC.Core.gp.gp_Trsf()
trsf.SetTranslation(OCC.Core.gp.gp_Vec(0,20,0))
thick_box_3 = OCC.Core.BRepBuilderAPI.BRepBuilderAPI_Transform(thick_box_2, trsf, True).Shape()

display.DisplayShape(thick_box)
display.DisplayShape(thick_box_3)
display.View.FitAll(0.08)
start_display()`
[image]

tpaviot, 2021-04-01:
Thank you for providing the full code, let me check what is wrong

tpaviot, 2021-04-01:
I confirm the issue with normals orientation. I'm not sure this is related to pythonocc itself, it may come from an issue with the occt kernel.

mirmik, 2021-04-02:
Thank you. I'll try to investigate where the problem occurs.

closed 2021-04-02 by mirmik