tpaviot/pythonocc-core #985: port to opencascade-7.5.1

tpaviot, 2021-03-30:
Follow-up of discussion #917 Refers to PR #984

The current master branch supports the latest 7.5.1 tag from upstream occt.

@looooo could you please report any feedback? Does the compilation pass on your machine? Is the conda-forge build process succesful?

looooo, 2021-03-30:
Can you create a new release which works with occt 7.5.1?

tpaviot, 2021-03-30:
ok, but I first need a few more checks

looooo, 2021-03-30:
ok, switched to master-branch to check:
https://github.com/conda-forge/pythonocc-core-feedstock/pull/4

tpaviot, 2021-03-30:
thanks for the pointer, I'll have a look at the build status before creating another release

tpaviot, 2021-03-30:
everything seems to be ok on the conda-forge side, I'm starting the release workflow

tpaviot, 2021-03-31:
Thank you @looooo for your help

closed 2021-03-31 by tpaviot