tpaviot/pythonocc-core #986: TFunction_DriverTable_AddDriver can't work

jfgajg, 2021-03-31:
Problem code:
        TFunction_DriverTable.Get().AddDriver(TOcafFunction_BoxDriver.GetID(), TOcafFunction_BoxDriver())

TOcafFunction_BoxDriver Inherited from TFunction_Driver, but Report an error:TypeError: in method 'TFunction_DriverTable_AddDriver', argument 3 of type 'opencascade::handle< TFunction_Driver > const &'

tpaviot, 2021-04-01:
Can you please provide a full sample ready to be executed without additional tweak, I can not duplicate the error

jfgajg, 2021-04-02:
driver.py:
##########################################################
from OCC.Core.TFunction import TFunction_Driver
from OCC.Core.TDataStd import TDataStd_Real, Handle_TDataStd_Real_Create
from OCC.Core.TNaming import TNaming_Builder
from OCC.Core.TPrsStd import TPrsStd_Driver
from OCC.Core.Standard import Standard_GUID, Standard_Transient
from OCC.Core.TFunction import TFunction_Logbook
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.gp import gp_Pnt
from OCC.Core._TFunction import TFunction_Driver_swigregister, TFunction_DriverTable_swigregister

class TOcafFunction_BoxDriver(TFunction_Driver):
    def __init__(self, *args):
        pass

    @staticmethod
    def GetID():
        return Standard_GUID("22D22E51-D63A-11d4-8F1A-0060B0EE18E8")

    def Validate(self, log, *args):
        log.SetValid(log.Label(), True)

    def MustExecute(self, log, *args):
        if log.IsModified(log.Label(), True):
            return True
        return False

    def Execute(self, log, *args):
        TSR = Handle_TDataStd_Real_Create()
        if not log.Label().FindChild(1).FindAttribute(TDataStd_Real.GetID(), TSR):
            return 1
        W = TSR.Get()
        if not log.Label().FindChild(2).FindAttribute(TDataStd_Real.GetID(), TSR):
            return 2
        L = TSR.Get()
        if not log.Label().FindChild(3).FindAttribute(TDataStd_Real.GetID(), TSR):
            return 3
        H = TSR.Get()
        if not log.Label().FindChild(4).FindAttribute(TDataStd_Real.GetID(), TSR):
            return 4
        X = TSR.Get()
        if not log.Label().FindChild(5).FindAttribute(TDataStd_Real.GetID(), TSR):
            return 5
        Y = TSR.Get()
        if not log.Label().FindChild(6).FindAttribute(TDataStd_Real.GetID(), TSR):
            return 6
        Z = TSR.Get()

        box = BRepPrimAPI_MakeBox(gp_Pnt(X, Y, Z), L, H, W)
        B = TNaming_Builder(log.Label())
        B.Generated(box.Shape())
        return 0

# TFunction_Driver_swigregister(TOcafFunction_BoxDriver)
# TFunction_DriverTable_swigregister(TOcafFunction_BoxDriver)

test.py:
###############################################
from OCC.Core.TFunction import TFunction_DriverTable
from OCC.Core.TDocStd import TDocStd_Application
from driver import TOcafFunction_BoxDriver

class myApp(TDocStd_Application):

    def __init__(self):
        super().__init__()
        TFunction_DriverTable.Get().AddDriver(TOcafFunction_BoxDriver.GetID(), TOcafFunction_BoxDriver())

if __name__ == '__main__':
    app = myApp()

##############################
Please execute test.py, You will see an error message.      version7.5.1

Thanks for your work.

tpaviot, 2021-04-02:
There's an issue with subclassing ``TFunction_Driver``. You have to properly instantiate the base classes:
```python
class TOcafFunction_BoxDriver(TFunction_Driver):

    def __init__(self, *args):
        super(TOcafFunction_BoxDriver, self).__init__()
```
The problem is that ``TFunction_Driver `` is an abstract class, as far as I understand

jfgajg, 2021-04-02:
This is the constructor of TFunction_Driverand cannot be executed.

class TFunction_Driver(OCC.Core.Standard.Standard_Transient):

    thisown = property(lambda x: x.this.own(), lambda x, v: x.this.own(v), doc="The membership flag")

    def __init__(self, *args, **kwargs):
        raise AttributeError("No constructor defined")

tpaviot, 2021-04-02:
Yep, because there are pure virtual methods inside the TFunction_Driver class

jfgajg, 2021-04-02:
> Yep, because there are pure virtual methods inside the TFunction_Driver class

What should I do?

tpaviot, 2021-04-02:
The ``TFunction_Driver`` class is a bit specific since it inherits from a ``Standard_Transient`` and defines virtual methods. That's why there's no construction defined and it behaves like an abstract class. ping @rainman110 
what you can do so far:
1. wait for a fix
2. pray for a quick fix

jfgajg, 2021-04-02:
> 2\. pray

Thanks for your answer.
In addition, I choose 2.

tpaviot, 2021-04-02:
Note, it's not a 7.5.1 specific issue, it's been there for years

jfgajg, 2021-04-02:
> Note, it's not a 7.5.1 specific issue, it's been there for years

It's just not resolved in the new version, so I posted this post.
So what is its repair process?
Need to write swig and compile it to fix it?
Or need to modify OpenCascade?
I don't know if I can solve it, but I am willing to try.