tpaviot/pythonocc-core #987: SetPixelFormat failed. Error code: 2000 raised from method Init of class Display3d

johrch, 2021-04-01:
hello. I just installed version 7.5.1. but I am getting this error when a run the core_classic_occ_bottle.py example.

(pyocc) C:\Projects\pythonocc-demos\examples>python core_classic_occ_bottle.py
creating bottle
bottle finished
qt-pyqt5 backend - Qt version 5.15.2
####### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
OpenGl_GraphicDriver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
WNT window created.
Traceback (most recent call last):
  File "core_classic_occ_bottle.py", line 210, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "F:\zapoye\anaconda3\envs\pyocc\lib\site-packages\OCC\Display\SimpleGui.py", line 198, in init_display
    win.canva.InitDriver()
  File "F:\zapoye\anaconda3\envs\pyocc\lib\site-packages\OCC\Display\qtDisplay.py", line 102, in InitDriver
    self._display.Create(window_handle=int(self.winId()), parent=self)
  File "F:\zapoye\anaconda3\envs\pyocc\lib\site-packages\OCC\Display\OCCViewer.py", line 176, in Create
    self.Init(self._window_handle)
  File "F:\zapoye\anaconda3\envs\pyocc\lib\site-packages\OCC\Core\Visualization.py", line 171, in Init
    return _Visualization.Display3d_Init(self, handle)
RuntimeError: Aspect_GraphicDeviceDefinitionErrorOpenGl_Window::CreateWindow: SetPixelFormat failed. Error code: 2000 raised from method Init of class Display3d

tpaviot, 2021-04-01:
This is a well-known issue, I finally managed to duplicate it on my machine. I found a fix, but I have to ensure it also works on your side. Can you please open the file ``F:\zapoye\anaconda3\envs\pyocc\lib\site-packages\OCC\Display\qtDisplay.py``, and change the line ``38`` from:

```python
class qtBaseViewer(QtOpenGL.QGLWidget):
```
to
```python
class qtBaseViewer(QtWidgets.QWidget):
```
Then restart the ``core_classic_occ_bottle_example``

tpaviot, 2021-04-01:
Related to #950

tpaviot, 2021-04-01:
Related to #911 at some point

johrch, 2021-04-01:
> This is a well-known issue, I finally managed to duplicate it on my machine. I found a fix, but I have to ensure it also works on your side. Can you please open the file [...]

its working now. thank you very much

tpaviot, 2021-04-01:
That's good news, thank you for the report! I will take care of closing this ticket when a fix is committed to the repository

tpaviot, 2021-04-01:
Related to #836

closed 2021-04-01 by tpaviot

tpaviot, 2021-05-11:
@metorm Yes it is merged into the master branch, but it has not been release yet.

It's a pythonocc problem related to the way it creates a graphic window in pyqt5 (pyqt4 is not supported anymore). This issue does not affect wxPython, not tested with pyside.

metorm, 2021-05-11:
Got it, thanks for your effort.

EmJay276, 2021-12-23:
> This is a well-known issue, I finally managed to duplicate it on my machine. I found a fix, but I have to ensure it also works on your side. Can you please open the file [...]

> @metorm Yes it is merged into the master branch, but it has not been released yet.
> 
> It's a pythonocc problem related to the way it creates a graphic window in pyqt5 (pyqt4 is not supported anymore). This issue does not affect wxPython, not tested with pyside.

This fix works with Pyside2 (5.13.2) and pythonocc-core 7.5.1

GeoGroup, 2022-01-01:
> 

My environment is Pyside2 (5.15.2) and pythonocc-core 7.5.1. This problem still exists by replacing class qtBaseViewer(QtWidgets.QWidget)

ManhoKANG, 2023-01-13:
This problem still exists.
My environment is window10, pyside2 and pythonocc-core 7.7.0.
I found that class qtBaseViewer(QtWidgets.QWidget): in line 33 of qtDisplay.py

commit a556a2db (2021-04-01): Replace QOpenGl with QWidget in qtDisplay.py Fix #987
commit ccce5b54 (2021-04-01): Replace QOpenGl with QWidget in qtDisplay.py Fix #987
commit a028f606 (2021-04-05): Replace QOpenGl with QWidget in qtDisplay.py Fix #987