tpaviot/pythonocc-core #988: Pyqt5 external 3D Viewer window

pridi, 2021-04-07:
Sorry if I post here but the Google group mailing list seems not working for me...
I'm trying to implement a small app in PyQt5. I have already created my main window with almost all the functionalities but now I would like to pop-up a new window containing the 3D model when a click a specific button. How can I accomplish this? The Simple.gui example creates again a main window and this is in conflict with my main window. Could you please help?

tpaviot, 2021-04-14:
This is more a PyQt5 issue I guess. You should not use the SimpleGui, but instanciate a qtViewer3d

Tanneguydv, 2021-04-14:
Hello,
here an example of PyQt5 integration of pythonocc with two buttons, one to display the Box, the other one to erase it. It's not a pop-up window (in a layout in the example) but I guess it could help with your question :

```
import sys
from PyQt5.QtWidgets import QApplication, QWidget, QPushButton, QHBoxLayout, QGroupBox, QDialog, QVBoxLayout
import OCC.Display.qtDisplay as qtDisplay
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

class App(QDialog):

    def __init__(self):
        super().__init__()
        self.title = 'PyQt5 / pythonOCC'
        self.left = 300
        self.top = 300
        self.width = 800
        self.height = 300
        self.initUI()

    def initUI(self):
        self.setWindowTitle(self.title)
        self.setGeometry(self.left, self.top, self.width, self.height)
        self.createHorizontalLayout()

        windowLayout = QVBoxLayout()
        windowLayout.addWidget(self.horizontalGroupBox)
        self.setLayout(windowLayout)
        self.show()

    def createHorizontalLayout(self):
        self.horizontalGroupBox = QGroupBox("Display PythonOCC")
        layout = QHBoxLayout()

        disp = QPushButton('Display Box', self)
        disp.clicked.connect(self.displayBOX)
        layout.addWidget(disp)

        eras = QPushButton('Erase Box', self)
        eras.clicked.connect(self.eraseBOX)
        layout.addWidget(eras)

        self.canvas = qtDisplay.qtViewer3d(self)
        layout.addWidget(self.canvas)
        self.canvas.resize(200, 200)
        self.canvas.InitDriver()
        self.display = self.canvas._display
        self.horizontalGroupBox.setLayout(layout)

    def displayBOX(self):
        a_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()
        self.ais_box = self.display.DisplayShape(a_box)[0]
        self.display.FitAll()

    def eraseBOX(self):
        self.display.Context.Erase(self.ais_box, True)

if __name__ == '__main__':
    app = QApplication(sys.argv)
    ex = App()
    sys.exit(app.exec_())

```

tpaviot, 2021-04-14:
Great @Tanneguydv I will add this contribution to the pythonocc-demos suite if you don't mind

tpaviot, 2021-04-14:
The example is here https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_qt5_app.py

I had to explicitly set the backend to 'qt-pyqt5'.

closed 2021-04-15 by tpaviot

pridi, 2021-04-15:
@Tanneguydv Thank you very very much! This is exactly what I needed!

Tanneguydv, 2021-04-15:
@tpaviot of course I don't mind, glad to be useful :)

@pridi Perfect!

pridi, 2021-04-16:
@Tanneguydv Sorry to bother you again. I'm trying to accomplish the offscreen render with the help of your code. When I try to request the Image Data with **data_640_480 =self.display.GetImageData(640, 480, Graphic3d_BufferType.Graphic3d_BT_RGB)** (as reported in the demos), I obtain a _None_ value for the variable data_640_480. This is quite strange because the self.display in your code, even if it is derived from qtBaseViewer, is a self.canvas._display that, in turn, is a OCCViewer.Viewer3d(), with all the properties and methods as the reported demo (GetImageData is a method of this class in fact). Do you have any suggestion?

Tanneguydv, 2021-04-16:
sorry I've never used this method, I'm quite new at using pythonocc... But if I get input I'll let you know

Tanneguydv, 2021-04-27:
To go further I've started to work on a didactic pythonocc modelisation/visualization tool implemented with PyQt5, you can find it here : 
https://github.com/Tanneguydv/poccad
It's a very humble start, as I'm new to all this

tpaviot, 2021-04-27:
Great!