tpaviot/pythonocc-core #989: Les fichiers core_webgl_xxxx.py et les autres

jmcornil, 2021-04-12:
Bonjour

J'ai installé ce week-end tout le *python-core* ainsi que les exemples. Joli travail, Merci !

Juste une petite remarque : à la fin de la partie "**Download/install binaries for Linux/OSX/Windows**", vous pourriez peut-être ajouter que tout cela permet de visualiser les fichier du type *core_webgl_xxxx.py* à l'aide d'un navigateur, mais que pour visualiser les autres, il est bon (indipensable ? ) d'installer dans l'environnement par exemple qt5 avec une instruction du type   
*conda install -c anaconda pyqt*
(en tout cas, c'est ce qui a marché pour moi)

Cordialement

Jack-Michel CORNIL

tpaviot, 2021-04-13:
qt, wxPython, pyside, numpy etc. can be used together with pythonocc, right, but they are not installed by default. qt is not necessary, there should be an explicit error message asking to install the missing dependencies. I will add a related paragraph to the README, to be even more explicit.

tpaviot, 2021-04-13:
J'ai donné ma réponse en anglais, c'est assez rare de voir du français sur cette plateforme de développement, mais ça fait du bien !

commit c24330c2 (2021-04-14): Small change to INSTALL.md, fix #989
closed 2021-04-14 by tpaviot