tpaviot/pythonocc-core #990: How to Dispaly point cloud like  Heatmap ?

aetzy, 2021-04-13:
I want to use Opencascade to dispaly point cloud with color ( _picture 1_) like a Heatmap (_picture 2,Geomagic Control Display_) .
However,I found that  setting the presentation attributes->point aspect property is not useful . Reference:[https://github.com/tpaviot/pythonocc-core/issues/744]
The point cloud size is about 350,000 points.if I use BRep spheres centered at each point, it's very very slow.
This has bothring me for days ,Thanks please.

[image]

[image]

aetzy, 2021-04-13:
@tpaviot  is there any method that can display point cloud like this?
Thank You !

tpaviot, 2021-04-13:
I think there is, but I don't know how to do it.

aetzy, 2021-04-15:
> 
> 
> I think there is, but I don't know how to do it.

Thank you for your time!
At the beginning，I want to Set the presentation attributes to make the points in cloud seems "Bigger".Code like this two lines:
`aPointCloud->Attributes()->PointAspect()->SetScale(10);`   
` myAISContext->DefaultDrawer()->PointAspect()->SetScale(10)`
but they didnot work.
If you have any ideas, please let me know.Or can you tell me if there are any similar examples？

Thanks a lot!