tpaviot/pythonocc-core #991: How can I find out camera coordinate (position)

b21604935, 2021-04-13:
After opening the step file, I can get the coordinates of each point on the shape. However, how can I learn the instant coordinates of the camera? 

[image]
[image]

tpaviot, 2021-04-13:
Have a look at the camera examples https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_camera_projection.py https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_camera_projection.py and https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_visualization_camera_2.py

b21604935, 2021-06-25:
Eye values do not change when I zoom in and out on the model. How can I generate new values?
I will be glad if you help.