tpaviot/pythonocc-core #993: Erase Single Part and not All

Tanneguydv, 2021-04-14:
Hello,

I want to Erase a shape from the display and the EraseAll method is not ok cause it erases all what's display.
For example if you want to keep another piece in your display it's not possible.
When I translate a part, the reference (`stl2` in that case), stays still,  :

```
trns = gp_Trsf()
trns.SetTranslation(gp_Vec(x_value, y_value, z_value))
moved = BRepBuilderAPI_Transform(stl2, trns, False).Shape()
display.DisplayShape(moved, update=True)
```
I guess that the `False` copy argument in the `BRepBuilderAPI_Transform` method concerns the framework and not the display.

I've seen that @henrikrudstrom has created an EraseShape method in 2011 using dictonnary and list.
tpaviot/pythonocc@ac560e9?branch=ac560e94fc088cece0ea6df5b49b26a2bbd3eb55&diff=unified
But I can not run his program

@tpaviot you responded to him "But why not create an AdvancedOCCViewer class, that inherits from OCCViewer, and provide additional and/or experimental features."
Is there a way to test simply this feature? It could be very useful for a lot of cases.

Thank you

tpaviot, 2021-04-14:
You can perform this action using the current pythonocc. The ``DisplayShape`` method returns a list that includes the ``AIS_Shapes``, just call the ``Erase`` method by passing this ``AIS_Shape`` instance.

```python
# display the shape
ais_box = display.DisplayShape(a_box)[0]
# remove the ais_shape from the viewer
display.Context.Erase(ais_box, True)
```

I pushed an example to the pythonocc-demos repository, see https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_erase_shape.py

I don't want to implement a mapping between TopoDS_Shapes and AIS_Shapes in the OCCViewer class, you can do it by yourself according to your needs.

closed 2021-04-14 by tpaviot

Tanneguydv, 2021-04-14:
Thank you very much for your quick and complete answer!