tpaviot/pythonocc-core #995: next release pythonocc-ore 7.5.2 and occt-7.5.2

tpaviot, 2021-04-26:
@looooo I noticed that a new occt version has been tagged (https://git.dev.opencascade.org/gitweb/?p=occt.git;a=commit;h=0dc2c377fc5a2d8cf065f4ec005e356240cb484c). Is there any plan at conda-forge to move occt up to 7.5.2?

looooo, 2021-04-26:
I read about some issues with that version:
https://forum.freecadweb.org/viewtopic.php?f=4&t=58090

tpaviot, 2021-04-26:
Interesting point, thank you.

tpaviot, 2021-04-26:
I would like to release a bugfix version for pythonocc-core, but if I upgrade the pythonocc version to 7.5.2, there might be confusion with the occt release. Any suggestion?

looooo, 2021-04-26:
7.5.1.1 ? not very convenient, but we did the same with smesh recently:
https://github.com/conda-forge/smesh-feedstock/blob/master/recipe/meta.yaml#L2

tpaviot, 2021-04-26:
That makes sense

looooo, 2021-05-13:
@tpaviot do we have to pin occt in the pythonocc-feedstock to the patch version of occt? Currently, only the minor version (x.x) is used for the pinning.

We will update to occt7.5.2 soon. So I guess you can release pythonocc also as 7.5.2.

tpaviot, 2021-05-16:
@looooo Ok, the current pythonocc-core release (7.5.1) should not be updated to 7.5.2

As soon as occt-7.5.2 is available at conda-forge, I will update pythonocc as well (there might be some API changes that require the python wrapper to be updated). I will notify you as soon as the move up to 7.5.2 is ready. Thank you for your help.

tpaviot, 2021-12-08:
There will not be any 7.5.2 version for pythonocc-core, let's move to 7.5.3 @looooo

tpaviot, 2021-12-08:
superseded by #1067

closed 2021-12-08 by tpaviot