tpaviot/pythonocc-core #1001: When I use BRepPrimAPI_MakeTorus in pythonocc to make a Torus, but I can't get the result I want.

VIC-LKH, 2021-05-10:
hello, thank you for providing a very nice python interface to OpenCasCade. When I used the BRepPrimAPI_MakeTorus function in pythonocc to make a Torus, I encountered a problem. Here is my code:

`from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeTorus
from OCC.Display.SimpleGui import init_display		
display, start_display, add_menu, add_function_to_menu = init_display()
from math import  pi

angle1 = -45
angle2 = 45
angle = 90
R1 = 5
R2 = 2

my_torus = BRepPrimAPI_MakeTorus(R1, R2, angle1/180*pi, angle2/180*pi, angle/180*pi).Shape()
display.DisplayShape(my_torus, update=True)
start_display()`

When the torus was displayed on my screen, the result was different from what I want. The result of code execution was just like this:
[image]

But what I want is just like this (the same as what has been described in Users' Guides in OpenCasCade Technology 7.5.0):
[image]

The version of pythonocc I use is 7.5.1 and Python is 3.7.10.  Is my understanding of Users' Guides wrong? or is there a bug in Pythonocc-7.5.1 code ? @tpaviot

tpaviot, 2021-05-10:
But I dont know why it fails

VIC-LKH, 2021-05-10:
I might use this function in this way, but I don't know why it fails too. If it is hard to find this bug, I will use another way to achieve the result I want.

tpaviot, 2021-05-10:
Hard to say if it's a pythonocc or occt issue

tpaviot, 2021-12-08:
Same result wih occt7.5.3. I think this is an occt bug which should be reported upstream.