tpaviot/pythonocc-core #1003: Getting part name and part origin location from assembly step file

iantferguson, 2021-05-12:
I have an assembly step file and I'm trying to return the part names and part origin locations with respect to the top-level assembly origin. I've tried something like this for getting origin location but all three parts come back as (0.0, 0.0, 0.0):

```
step_reader = STEPControl_Reader()
step_reader.ReadFile('./test_assy.stp')
step_reader.TransferRoots()
myshape = step_reader.Shape()

it = TopoDS_Iterator(myshape)
while it.More():
   shp = it.Value()
   trsf = myshape.Location().Transformation()
   print (trsf.TranslationPart().Coord())
   it.Next()
```

I'm also not sure if it's more appropriate to use XCAF and something like ShapeTool which I've seen in some assembly examples.

iantferguson, 2021-05-13:
Found my answer by looking at DataExchange.py: https://github.com/tpaviot/pythonocc-core/blob/master/src/Extend/DataExchange.py

tpaviot, 2021-05-14:
I reopen the issue, I have additional material related to this topic