tpaviot/pythonocc-core #1004: Part naming logic via DataExchange

paddy-r, 2021-05-17:
Thomas, can you comment on the logic for string parsing when reading files? I have a part with an unusual name containing multiple apostrophes that are all removed except one. Specifically, the string contains "C''''''''SK" (that's eight apostrophes) in the PRODUCT and SHAPE_REPRESENTATION lines that is parsed to "C'SK".

I couldn't find any reference to this in the PythonOCC or Open Cascade docs but apologies if I've missed it.

paddy-r, 2021-05-19:
Out of interest, there is a patch that Kirill mentions [here](https://dev.opencascade.org/content/step-item-name-parsing#comment-22154), as of OCCT 7.5.0.

tpaviot, 2021-05-31:
ok, I close the issue.

closed 2021-05-31 by tpaviot

paddy-r, 2021-06-04:
Hi Thomas, the problem seems to still be present, so I don't know whether you want to reopen the issue. I have made a note at the Open Cascade forum [here](https://dev.opencascade.org/content/step-item-name-parsing) anyway. However, I realise it's a fringe issue!