tpaviot/pythonocc-core #1009: The problem with reading step

dayhappy2012, 2021-05-31:
Other than reading local files, is there any way to read binary? thanks

tpaviot, 2021-05-31:
I dont' understand the question, can you please develop

dayhappy2012, 2021-06-01:
STEPCAFControl_Reader

StepCaFControl_Reader has only the readFile attribute (file path),

But I have the binary data of the file

Is there a way to accept the data directly?

tpaviot, 2021-06-01:
Thank you for the clarification, but I still don't understand I'm sorry. What do you call the *binary date of the file*?

dayhappy2012, 2021-06-01:
Such as:

    step_reader = STEPCAFControl_Reader()

    with open('a.stp', 'rb') as e:
        my_object = e.read()
        step_reader.read_object(my_object )

But my my_object comes from somewhere else and is not a read file

tpaviot, 2021-06-01:
I'm sorry but I still didn't get it. I leave the issue open, if ever someone coming around can give you any clue.

leycec, 2021-06-29:
Sure! I'll be someone. :smile:

What @dayhappy2012 is asking is how to read a binary stream of in-memory bytes (e.g., `bytes`, `bytearray`) comprising a STEP file. They don't actually *have* a physical on-disk STEP file; they only have the abstract in-memory contents of what would be a STEP file if they actually had one, which they don't.

Is that about right, @dayhappy2012? My advice for the moment would be just to write your byte stream to a temporary file and read from that. Yes, that's lame, but sometimes you just gotta be lame and ship that code out.

dayhappy2012, 2021-07-30:
Export assembly structure tree, what method to export STP format, do you have demo? @leycec @tpaviot

closed 2021-12-08 by tpaviot