tpaviot/pythonocc-core #1010: How to judge that two TopoDS_Shape objects are the same

luoyu-123, 2021-06-03:
I generate two entities of the same TopoDS_Shape, the parameters of generating them are the same, how can I judge that they are the same, I use the TopoDS_Shape class
IsEqual attribute and IsSame attribute but return false

thank you

ts0ron, 2021-07-25:
Hey,

I am a bit new to OCC but it seems that when creating 2 shapes with the same properties they still has 2 different identifier.
A shape is a compound object and not a number such as (2 =2 ) so I guess when creating a new shape it has a unique identifier and maybe that's why the False result.

Have you tried to compare their boundaries and the face-equation ?

tpaviot, 2021-12-04:
@luoyu-123 can you please insert a code snippet