tpaviot/pythonocc-core #1015: No module named 'OCC.Core.MeshDS'

ShieKn, 2021-06-18:
HI, Thanks for the pythonocc
I have to display stl files with about 200Mb. it's very slow when i use ‘read_stl_file()’ functions. i found this(#992 )about
efficiency of importing mesh data, but some module like 'OCC.Core.MeshDS' can't import when i run the example named core_mesh_data_source.py, neither version 7.5.1 nor version 7.4.0rc1 works. How should I install this extra MeshDS？
Thank you very much for your help and ideas.

tpaviot, 2021-12-03:
is not available in pythonocc-core-7.5.1, will be released in the next public version

closed 2021-12-03 by tpaviot

VoladorZou, 2022-07-08:
[image]
Hi, Thanks for the pythonocc
May I know if you have any plans to release a new version this year? pythonocc-core-7.5.1 was released one year ago.
'OCC.Core.MeshDS' is one important new feature for us.

VoladorZou, 2022-07-08:
> is not available in pythonocc-core-7.5.1, will be released in the next public version

Hi, Thanks for the pythonocc
May I know if you have any plans to release a new version this year? pythonocc-core-7.5.1 was released one year ago.
'OCC.Core.MeshDS' is one important new feature for us.

VoladorZou, 2022-07-08:
> is not available in pythonocc-core-7.5.1, will be released in the next public version
In additon, we want to know the performance of  'OCC.Core.MeshDS' .
for example, how much time it will take when read one 200M STL flie.