tpaviot/pythonocc-core #1018: Fusing Faces of a Shell

arbrasington, 2021-07-01:
I am interested in working in the u and v parameter space of a shell. However, this space is not continuous when multiple faces are present. Each face has its own local u and v coordinate representation. For example, 2 faces can both have u and v ranges from 0 to 1 instead of one face being from 0 to 1 and the next being from 1 to 2. I haven't seen any way to do this with the shell itself. My next thought is to fuse the faces of the shell together to create one common face. 

Has anybody been able to accomplish this? I've tried BRepLib_FuseEdges and ShapeUpgrade_UnifySameDomain with no luck. Multiple faces still persist after trying these. I've also tried reorganizing the uv parameters based on face locations, but this is nearly impossible with a lot of faces or oddly shaped faces.

Any ideas are appreciated! I can provide some example shells if desired.

tpaviot, 2021-12-06:
Fusing the faces into a compound won't solve the issue. IMO you have to approximate the set of faces using a single surface, but you will loose all inner edges

Tanneguydv, 2021-12-06:
I would be interested to take a look at an example shell

arbrasington, 2022-07-13:
[spline_faces.zip](https://github.com/tpaviot/pythonocc-core/files/9102949/spline_faces.zip)

An example stp file is in the zip file.