tpaviot/pythonocc-core #1025: LNK2019: unresolved external symbol

seewjr, 2021-08-03:
Hi, I have been trying to build the pythonocc from source code.

The platform is Win10, compiler is msvc from vs2019 community and I have tried to build both the pythonocc 7.5.1 version and 7.5.0rc1 version but they all failed

First I successfully build the opencscade 7.5.1 and opencascade 7.5.0, and then I followed the instruction in install.md to build pythonocc 7.5.1 and pythonocc 7.5.0rc1 , the swig version is 3.0.11. The erros are same, after building All_Build in vs2019 there are many LNK2019 erros, one of them is paste here:

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "public: __cdecl TDF_Transaction::TDF_Transaction(class TCollection_AsciiString const &)" (??0TDF_Transaction@@QEAA@AEBVTCollection_AsciiString@@@Z) referenced in function "public: void __cdecl TDF_Transaction::`default constructor closure'(void)" (??_FTDF_Transaction@@QEAAXXZ)	_IGESCAFControl	C:\Googol_Projects\FREECAD0.2\PYTHONOCC751\pythonocc-core-751-build\IGESCAFControlPYTHON_wrap.obj	1	

I have 287 projects in pythonocc successfully built but 17 failed. There are many LNK2019 erros. How should I resolve this issue? Thank you very much!

ziyeye, 2021-08-12:
The way that works for me is to change the project to be "static library" instead of "dynamic dll" in visual studio by right clicking on the project.

tpaviot, 2021-08-24:
This kind of errors usually comes from a conflict between pythonocc-core required occt version and the occt version installed on your machine

closed 2021-12-06 by tpaviot

wanghaoyang1995, 2022-10-26:
> Hi, I have been trying to build the pythonocc from source code.
>
> The platform is Win10, compiler is msvc from vs2019 community and I have tried to build both the pythonocc 7.5.1 version and 7.5.0rc1 [...]

I have met this problem, similarly with 286 succeeded and 17 failed.
(My occ version is 7.6.2, which is compatible with pythonocc-core7.6.2.)
Is this the mistakes of pythonocc-core?

The follow is the error log:
```
Build started...
1>------ Build started: Project: _XmlXCAFDrivers, Configuration: RelWithDebInfo x64 ------
2>------ Build started: Project: _XmlMXCAFDoc, Configuration: RelWithDebInfo x64 ------
3>------ Build started: Project: _XCAFPrs, Configuration: RelWithDebInfo x64 ------
4>------ Build started: Project: _XCAFDoc, Configuration: RelWithDebInfo x64 ------
5>------ Build started: Project: _XCAFDimTolObjects, Configuration: RelWithDebInfo x64 ------
6>------ Build started: Project: _XCAFApp, Configuration: RelWithDebInfo x64 ------
7>------ Build started: Project: _VrmlData, Configuration: RelWithDebInfo x64 ------
8>------ Build started: Project: _VrmlConverter, Configuration: RelWithDebInfo x64 ------
9>------ Build started: Project: _VrmlAPI, Configuration: RelWithDebInfo x64 ------
10>------ Build started: Project: _TPrsStd, Configuration: RelWithDebInfo x64 ------
11>------ Build started: Project: _STEPCAFControl, Configuration: RelWithDebInfo x64 ------
12>------ Build started: Project: _RWObj, Configuration: RelWithDebInfo x64 ------
13>------ Build started: Project: _RWMesh, Configuration: RelWithDebInfo x64 ------
14>------ Build started: Project: _RWGltf, Configuration: RelWithDebInfo x64 ------
1>   Creating library D:/Files/work_temp/pythonocc-core-7.6.2/build/bin/RelWithDebInfo/_XmlXCAFDrivers.lib and object D:/Files/work_temp/pythonocc-core-7.6.2/build/bin/RelWithDebInfo/_XmlXCAFDrivers.exp
1>XmlXCAFDriversPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_Transaction::TDF_Transaction(class TCollection_AsciiString const &)" (??0TDF_Transaction@@QEAA@AEBVTCollection_AsciiString@@@Z) referenced in function "public: void __cdecl TDF_Transaction::`default constructor closure'(void)" (??_FTDF_Transaction@@QEAAXXZ)
1>XmlXCAFDriversPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_ClosureMode::TDF_ClosureMode(bool)" (??0TDF_ClosureMode@@QEAA@_N@Z) referenced in function "public: void __cdecl TDF_ClosureMode::`default constructor closure'(void)" (??_FTDF_ClosureMode@@QEAAXXZ)
1>XmlXCAFDriversPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_IDFilter::TDF_IDFilter(bool)" (??0TDF_IDFilter@@QEAA@_N@Z) referenced in function "public: void __cdecl TDF_IDFilter::`default constructor closure'(void)" (??_FTDF_IDFilter@@QEAAXXZ)
1>XmlXCAFDriversPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_RelocationTable::TDF_RelocationTable(bool)" (??0TDF_RelocationTable@@QEAA@_N@Z) referenced in function "public: void __cdecl TDF_RelocationTable::`default constructor closure'(void)" (??_FTDF_RelocationTable@@QEAAXXZ)
1>D:\Files\work_temp\pythonocc-core-7.6.2\build\bin\RelWithDebInfo\_XmlXCAFDrivers.pyd : fatal error LNK1120: 4 unresolved externals
1>Done building project "_XmlXCAFDrivers.vcxproj" -- FAILED.
15>------ Build started: Project: _IGESCAFControl, Configuration: RelWithDebInfo x64 ------
2>   Creating library D:/Files/work_temp/pythonocc-core-7.6.2/build/bin/RelWithDebInfo/_XmlMXCAFDoc.lib and object D:/Files/work_temp/pythonocc-core-7.6.2/build/bin/RelWithDebInfo/_XmlMXCAFDoc.exp
2>XmlMXCAFDocPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_ClosureMode::TDF_ClosureMode(bool)" (??0TDF_ClosureMode@@QEAA@_N@Z) referenced in function "public: void __cdecl TDF_ClosureMode::`default constructor closure'(void)" (??_FTDF_ClosureMode@@QEAAXXZ)
2>XmlMXCAFDocPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_IDFilter::TDF_IDFilter(bool)" (??0TDF_IDFilter@@QEAA@_N@Z) referenced in function "public: void __cdecl TDF_IDFilter::`default constructor closure'(void)" (??_FTDF_IDFilter@@QEAAXXZ)
2>XmlMXCAFDocPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_RelocationTable::TDF_RelocationTable(bool)" (??0TDF_RelocationTable@@QEAA@_N@Z) referenced in function "public: void __cdecl TDF_RelocationTable::`default constructor closure'(void)" (??_FTDF_RelocationTable@@QEAAXXZ)
2>XmlMXCAFDocPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_Transaction::TDF_Transaction(class TCollection_AsciiString const &)" (??0TDF_Transaction@@QEAA@AEBVTCollection_AsciiString@@@Z) referenced in function "public: void __cdecl TDF_Transaction::`default constructor closure'(void)" (??_FTDF_Transaction@@QEAAXXZ)
2>D:\Files\work_temp\pythonocc-core-7.6.2\build\bin\RelWithDebInfo\_XmlMXCAFDoc.pyd : fatal error LNK1120: 4 unresolved externals
2>Done building project "_XmlMXCAFDoc.vcxproj" -- FAILED.
3>   Creating library D:/Files/work_temp/pythonocc-core-7.6.2/build/bin/RelWithDebInfo/_XCAFPrs.lib and object D:/Files/work_temp/pythonocc-core-7.6.2/build/bin/RelWithDebInfo/_XCAFPrs.exp
3>XCAFPrsPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: class TDF_Label const __cdecl TDF_Attribute::Label(void)const " (?Label@TDF_Attribute@@QEBA?BVTDF_Label@@XZ) referenced in function "public: class TDF_Label __cdecl XCAFDoc_VisMaterialTool::BaseLabel(void)const " (?BaseLabel@ [... cut ...] saction::`default constructor closure'(void)" (??_FTDF_Transaction@@QEAAXXZ)
17>D:\Files\work_temp\pythonocc-core-7.6.2\build\bin\RelWithDebInfo\_BinMXCAFDoc.pyd : fatal error LNK1120: 4 unresolved externals
17>Done building project "_BinMXCAFDoc.vcxproj" -- FAILED.
18>------ Skipped Build: Project: INSTALL, Configuration: RelWithDebInfo x64 ------
18>Project not selected to build for this solution configuration 
========== Build: 0 succeeded, 17 failed, 286 up-to-date, 1 skipped ==========

```

wanghaoyang1995, 2022-10-26:
Hi friend! I have found the reason. 

This is because those OCC module lack some xxx.lib. (Maybe the writer's mistakes)

You need to find those  error modules need what libs file, and fix the missing link.

For example:

This is my error log:
```
Creating library D:/Files/work_temp/pythonocc-core-7.6.2/build/bin/Debug/_XmlXCAFDrivers.lib and object D:/Files/w
  ork_temp/pythonocc-core-7.6.2/build/bin/Debug/_XmlXCAFDrivers.exp
XmlXCAFDriversPYTHON_wrap.obj : error LNK2019: unresolved external symbol "public: __cdecl TDF_Transaction::TDF_Transac
tion(class TCollection_AsciiString const &)" (??0TDF_Transaction@@QEAA@AEBVTCollection_AsciiString@@@Z) referenced in f
unction "public: void __cdecl TDF_Transaction::`default constructor closure'(void)" (??_FTDF_Transaction@@QEAAXXZ) [D:\
Files\work_temp\pythonocc-core-7.6.2\build\_XmlXCAFDrivers.vcxproj]
...
```

Just to modify the CMakeList.txt, add the dependency `TKLCAF` to those lines. (as the bold font)

> set(OCE_MODEL_LIBRARIES TKernel TKMath TKBRep TKG2d TKG3d TKGeomBase TKBO
>                         TKBool TKFeat TKFillet TKGeomAlgo TKHLR TKMesh TKOffset
>                         TKPrim TKShHealing TKTopAlgo TKXMesh)
> set(OCE_VISUALIZATION_LIBRARIES TKService TKV3d TKOpenGl TKMeshVS)
> set(OCE_DATAEXCHANGE_LIBRARIES TKBinXCAF TKIGES **TKLCAF** TKRWMesh TKSTEP TKSTEP209 TKSTEPAttr
>                                TKSTEPBase TKSTL TKVRML TKXCAF TKXDEIGES TKXDESTEP TKXSBase
>                                TKXmlXCAF)
> set(OCE_OCAF_LIBRARIES TKCDF TKLCAF TKCAF TKBinL TKXmlL TKBin TKXml TKStdL
>                        TKStd TKTObj TKBinTObj TKXmlTObj TKVCAF)

wanghaoyang1995, 2022-10-27:
Final changes:
```cmake
#############################################
# List of OCE shared libraries to link with #
#############################################
set(OCE_MODEL_LIBRARIES TKernel TKMath TKBRep TKG2d TKG3d TKGeomBase TKBO
                        TKBool TKFeat TKFillet TKGeomAlgo TKHLR TKMesh TKOffset
                        TKPrim TKShHealing TKTopAlgo TKV3d TKXMesh)
set(OCE_VISUALIZATION_LIBRARIES TKService TKV3d TKOpenGl TKMeshVS)
set(OCE_DATAEXCHANGE_LIBRARIES TKBinXCAF TKCDF TKLCAF TKIGES TKRWMesh TKService TKSTEP TKSTEP209 TKSTEPAttr
                               TKSTEPBase TKSTL TKVRML TKVCAF TKXCAF TKXDEIGES TKXDESTEP TKXSBase
                               TKXmlXCAF)
set(OCE_OCAF_LIBRARIES TKCDF TKLCAF TKCAF TKBinL TKXmlL TKBin TKXml TKService TKStdL
                       TKStd TKTObj TKBinTObj TKXmlTObj TKVCAF)

```

========== Build: 303 succeeded, 0 failed, 0 up-to-date, 1 skipped ==========

tsvilans, 2023-02-27:
I noticed today that I also had to add `TKV3d` to `OCE_DATAEXCHANGE_LIBRARIES` and `OCE_OCAF_LIBRARIES` and the include directory for `rapidjson` needed to be set in `_RWGltf.vcproj`, then it seems to work :)