tpaviot/pythonocc-core #1026: PythonOCC says No module named 'OCC.STEPControl'.

AmanSavaria1402, 2021-08-08:
Hello,

I am trying to extract features from step files by following [this](https://github.com/tpaviot/pythonocc-core/issues/470#issuecomment-348202128) commit.
But when I import the required functions,
`from OCC.STEPControl import STEPControl_Reader
from OCC.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCC.Display.SimpleGui import init_display
from OCC.BRep import BRep_Tool_Surface
from OCC.GeomAbs import GeomAbs_Plane, GeomAbs_Cylinder
from OCC.TopoDS import topods_Face
from OCC.BRepAdaptor import BRepAdaptor_Surface`

I get a `ModuleNotFoundError` in importing all these functions except for the `From OCC.Display.SimpleGui import init_display` function.

These are all the error messages:
`
ModuleNotFoundError: No module named 'OCC.STEPControl'
ModuleNotFoundError: No module named 'OCC.IFSelect'
ModuleNotFoundError: No module named 'OCC.BRep'
ModuleNotFoundError: No module named 'OCC.GeomAbs'
ModuleNotFoundError: No module named 'OCC.TopoDS'
ModuleNotFoundError: No module named 'OCC.BRepAdaptor'
`

Currently, I am using pythonocc-core == 7.5.1 and occt==7.5.2

Thank You.

mkskalai, 2021-08-12:
Hello,

try using OCC.Core.STEPControl, OCC.Core.BRep etc.

Hope this helps

closed 2021-08-24 by tpaviot

msm-prarthana-bataju, 2022-06-23:
Hello @mkskalai,
I am sorry but I didn't understand your answer. Could you please briefly write it.
I am also facing the same dll error and tried every possible steps but didn't work. :(