tpaviot/pythonocc-core #1029: Node selection using meshVS

ferhubsch, 2021-08-13:
Hello there! I appreciate if someone could help me with an issue that I'm having while using meshVS.

**Goal**: I wish to allow the user to select nodes displayed in a meshVS. 

**Problem**: I cannot use OCCViewer's register_select_callback method because the method OCCViewer.Select looks for shapes (using the Context.SelectedShapes method as shown below) and meshVS objects (as far as I know) cannot be converted easily to AIS_Shape objects. I am pretty sure my mesh object has the right selection flags, since I manage to detect that there is a single node being detected after a click (self.Context.NbSelected() method gives =1 as output), but I cannot easily find the node's coordinates within 

```
def Select(self, X, Y):
        self.Context.Select(True)
        self.Context.InitSelected()
        self.selected_shapes = []

        if self.Context.MoreSelected():   # I get a True return here
            if self.Context.HasSelectedShape(): # But here is where I fail to extract the nodes information
                self.selected_shapes.append(self.Context.SelectedShape())

        for callback in self._select_callbacks:
            callback(self.selected_shapes, X, Y)
```

**What I tried to do**: I only managed to find a few objects (as follows) that don't have any information on the nodes  coordinates or on the meshVS object itself.
```
AIS_Selection_Object= self.Context.Selection()
SelectMgr_EntityOwner = AIS_Selection_Object.Objects().First()
AIS_NListOfEntityOwner = AIS_Selection_Object.Objects()
```

**Environment**: I am currently using occ version 7.5.2, python 3.7.10

tpaviot, 2021-08-24:
Can you please provide a description of the issue.

ferhubsch, 2021-08-24:
Hello Sir! I edited the description and added a piece of code, hopping for any insights on the matter. I thank you for any help, if possible.

zjuzhh, 2023-05-18:
Hello sir ,
    I meet the same question.
    I ues v3d_view to display some cad model(ais_shape) and pointcloud model(ais_pointcloud)succesfully, but when i want to select some points from an ais_pointcloud,it's not the same as selecting TopAbs_Face from the AIS_Slection.
    The schematic diagram is as follows:
[image]
    The code I select TopAbs_Face from an ais_shape object is as follows:
```       
   if (!myOccView->getContext()->Selection()->IsEmpty()) {
        const Handle(AIS_Selection) selection = myOccView->getContext()->Selection();
        for (selection->Init(); selection->More(); selection->Next()) {
            Handle(SelectMgr_EntityOwner) entity = selection->Value();
            TopoDS_Shape shape = Handle(StdSelect_BRepOwner)::DownCast(entity)->Shape();

            // Create an explorer to iterate over all faces in the shape.
            TopExp_Explorer explorer_face(shape, TopAbs_FACE);
            ......
            }
    }
```
     The code I select points from an ais_shape object is as follows:
```
       Handle(AIS_Selection) selection = myOccView->getContext()->Selection();       
       Handle(AIS_PointCloud) myPointCloud = Handle(AIS_PointCloud)::DownCast(aisObj);
       for (selection->Init(); selection->More(); selection->Next()) {
                Handle(SelectMgr_EntityOwner) entity = selection->Value();
                Handle(AIS_PointCloudOwner) owner = Handle(AIS_PointCloudOwner)::DownCast(entity);
                Handle(TColStd_HPackedMapOfInteger) selectedPoints = owner->SelectedPoints();
                ......
       }
```
     I don't how to use the selectedPoints(TColStd_HPackedMapOfInteger object) to get every gp_pnt point，could you give me some advice or some sample code？Thanks in advance.