tpaviot/pythonocc-core #1031: Retrieve Curve from BRepAlgoAPI_Section result

Tanneguydv, 2021-08-24:
Hello,
I don't get the way we could get the result of an intersection of a Plane with a Shape as a Curve, I believe we need to translate the result of BRepAlgoAPI_Section into a curve but I can't find how...

Could you please help me on that point?

As an example here one of the demos code  :
[https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_boolean.py](url)
```
def slicer(event=None):
    # Param
    Zmin, Zmax, deltaZ = -100, 100, 5
    # Note: the shape can also come from a shape selected from InteractiveViewer
    if 'display' in dir():
        shape = display.GetSelectedShape()
    else:
        # Create the shape to slice
        shape = BRepPrimAPI_MakeSphere(60.).Shape()
    # Define the direction
    D = gp_Dir(0., 0., 1.)  # the z direction
    # Perform slice
    sections = []
    init_time = time.time()  # for total time computation
    for z in range(Zmin, Zmax, deltaZ):
        # Create Plane defined by a point and the perpendicular direction
        P = gp_Pnt(0, 0, z)
        Pln = gp_Pln(P, D)
        face = BRepBuilderAPI_MakeFace(Pln).Shape()
        # Computes Shape/Plane intersection
        section_shp = BRepAlgoAPI_Section(shape, face)
        if section_shp.IsDone():
            sections.append(section_shp)
    total_time = time.time() - init_time
    print("%.3fs necessary to perform slice." % total_time)

    display.EraseAll()
    display.DisplayShape(shape)
    for section_ in sections:
        display.DisplayShape(section_.Shape())
    display.FitAll()
```

Kri-Ol, 2022-08-03:
Hi, did you solve the issue? I have somewhat similar problem - Z-axis symmetric shape (but not done by rotation, relatively large STEP file), and would like to cut it using, say ZY (or ZX) plane, and get cut-off curve as set of Z,Y points. Any advices are greatly appreciated

qigong777, 2022-08-03:
> Hi, did you solve the issue? I have somewhat similar problem - Z-axis symmetric shape (but not done by rotation, relatively large STEP file), and would like to cut it using, say ZY (or ZX) plane, and get cut-off curve as set of Z,Y points. Any advices are greatly appreciated

I used the above function TopoDS_Iterator, but the curve obtained is quite different from the expected, and I don't know how to successfully implement the data transform.

Kri-Ol, 2022-08-04:
@qigong777  Thank you. I just implemented the same logic, and got bunch of edges (of type TopoDS_Edge), and they together looks right in the UI/viewer! So my next question is - could I get just bunch of points from each edge? Is there somewhere hidden parametric representation of the edge? Should be, because how else they are drawn on the screen?

qigong777, 2022-08-04:
> @qigong777 Thank you. I just implemented the same logic, and got bunch of edges (of type TopoDS_Edge), and they together looks right in the UI/viewer! So my next question is - could I get just bunch [...]

I have the same problem. If I display the transformed edges , the two edges coincide exactly, but if I convert them to the coordinates at the head of several edge, the display is incorrect. I am still trying to figure out the problem. And my code is as follows:

line = BRepAlgoAPI_Section(stlshapes, rectShape1.Shape(), True)
    display, start_display, add_menu, add_function_to_menu = init_display()
    it = TopoDS_Iterator(line.Shape(), True, True)
    edge = it.Value()
    print(type(edge))

    edges = []
    while it.More():
        edges.append(it.Value())
        it.Next()
    mycurve = TColgp_HArray1OfPnt(1, len(edges))
    PTS = []
    for i, edge in enumerate(edges):
        stPoint = BRep_Tool.Curve(edge)[0]
        pt = [stPoint.Value(0).X(), stPoint.Value(0).Y(), stPoint.Value(0).Z()]
        PTS.append(pt)
        PTS = sorted(PTS, key=lambda a: a[0])
    for j, po in enumerate(PTS):
        mycurve.SetValue(j+1, gp_Pnt(po[0],po[1],po[2]))
        display.DisplayShape(gp_Pnt(po[0],po[1],po[2]), update=True)
    mycurve = GeomAPI_Interpolate(mycurve, False, 0.01)
    mycurve.Perform()
    display.DisplayShape(mycurve.Curve())

    #display.DisplayColoredShape(edge, "YELLOW")
    display.DisplayShape(line.Shape(), update=True, color=rgb_color(0, 0, 1))

    #display.FitAll()
    start_display()

Do you have any good suggestions?

qigong777, 2022-08-04:
> @qigong777 Thank you. I just implemented the same logic, and got bunch of edges (of type TopoDS_Edge), and they together looks right in the UI/viewer! So my next question is - could I get just bunch [...]

You can extract the starting point of each edge. The code is shown below：

    line = BRepAlgoAPI_Section(stlshapes, rectShape2.Shape(), True)
    display, start_display, add_menu, add_function_to_menu = init_display()
    it = TopoDS_Iterator(line.Shape(), True, True)
    edges = []
    while it.More():
        edges.append(it.Value())
        it.Next()
    for i, edge in enumerate(edges):
        [curve_handle, a, b] = BRep_Tool.Curve(edge)
        pt = [curve_handle.Value(a).X(), curve_handle.Value(a).Y(), curve_handle.Value(a).Z()]
        display.DisplayShape(gp_Pnt(pt[0],pt[1],pt[2]), update=True, color=rgb_color(1, 0, 0))
        #display.DisplayColoredShape(edges[j], "YELLOW")
    display.DisplayShape(line.Shape(), update=True, color=rgb_color(0, 0, 1))
    display.FitAll()
    start_display()

[image]