tpaviot/pythonocc-core #1033: Possibility of suppressing print messages using Transfer method of STEPControl_Writer?

Krande, 2021-08-30:
Hi,

I am just wondering if it is possible to suppress the print messages when using the `Transfer` method of `STEPControl_Writer`?
In my particular use case I transfer a large number of shapes, and having the ability to suppress the printouts would be great! An alternative solution would be if there are any alternatives of transferring shapes to `STEPControl_Writer` that would not result in a print statement for each shape?

I am using the following versions of pythonocc-core and occt:
```
occt		7.5.1            h60997fb_2        conda-forge
pythonocc-core	7.5.1            py39h5d9d39a_0    conda-forge
```

A minimum example from [pythonocc-demos](https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_export_step_ap203.py):

```
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

from OCC.Core.STEPControl import STEPControl_Writer, STEPControl_AsIs
from OCC.Core.Interface import Interface_Static_SetCVal
from OCC.Core.IFSelect import IFSelect_RetDone

# creates a basic shape
box_s = BRepPrimAPI_MakeBox(10, 20, 30).Shape()

# initialize the STEP exporter
step_writer = STEPControl_Writer()
Interface_Static_SetCVal("write.step.schema", "AP203")

# transfer shapes and write file
step_writer.Transfer(box_s, STEPControl_AsIs)
status = step_writer.Write("box.stp")

if status != IFSelect_RetDone:
	raise AssertionError("load failed")
```
The print message it creates for each `step_writer.Transfer(..)` operation
```
*******************************************************************
******        Statistics on Transfer (Write)                 ******

*******************************************************************
******        Transfer Mode = 0  I.E.  As Is       ******
******        Transferring Shape, ShapeType = 2                      ******
```

Let me know if you need anything else to help advance this issue!

Best Regards
Kristoffer

cafhach, 2021-10-20:
I think this is not possible at the moment. There are two ways in which this could work but they require changes to the SWIG wrapper code:

a) You could alter the `Message::DefaultMessenger`. This is a static object which is probably used by STEP_Writer if you not tell it explicitely to use something else and on this singleton object you could just remove the MessagePrinter entry for stdout
b) In your example above, one should be able to access the active messenger for the write operation using `step_writer.WS().TransferWriter().FinderProcess().Messenger()` but `Messenger()` is not wrapped yet.

tpaviot, 2021-12-06:
@cafhach The ```Message_Messenger``` class wrapper is available in the current master branch
```bash
>>> from OCC.Core.Message import *
>>> Message_Messenger
<class 'OCC.Core.Message.Message_Messenger'>
>>> Message_Messenger()
<class 'Message_Messenger'>
>>> a = Message_Messenger()
>>> dir(a)
['AddPrinter', 'ChangePrinters', 'DecrementRefCounter', 'Delete', 'DownCast', 'DumpJsonToString', 'DynamicType', 'GetRefCount', 'IncrementRefCounter', 'IsInstance', 'IsKind', 'Printers', 'RemovePrinter', 'RemovePrinters', 'Send', 'SendAlarm', 'SendFail', 'SendInfo', 'SendTrace', 'SendWarning', 'This', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__swig_destroy__', '__weakref__', 'get_type_descriptor', 'get_type_name', 'this', 'thisown']
```

andreas-wolf, 2021-12-09:
I have the same problem with read_step_file. I try to work on a batch of files and it floods the console with unwanted status output. 
I assume from a comment above about Message::DefaultMessenger being a singelton, that it might be involved in the output of read_step_file too.
Maybe it is possible to send the output to a python logger (which we then can mute via the loglevel)?