tpaviot/pythonocc-core #1050: How to accelerate the function defined by myself with multiprocessing python-module in PythonOCC?

VIC-LKH, 2021-11-09:
hello, thank you for providing a very nice python interface to OpenCasCade. Now I have many points (gp_Pnt) and want to use class <BRepClass3d_SolidClassifier> to judge their relative positions to the geometries defined in my code. I want to use multiprocessing or concurrent.futures Python-module to accelerate this process. However, it says  "can‘t pickle SwigPyObject objects", which means the objects are not serializable. What should I do if I want to accelerate this task with multi-CPUs? @tpaviot

tpaviot, 2021-12-01:
You have to implement your own serialization functions if you want to pickle objects so that they can be shared between processes. Multiprocessing works quite well, you can have a look at https://github.com/tpaviot/pythonocc-demos/blob/256079d3104dfbb1f7f9fe244bdbd66c916739ef/examples/core_parallel_slicer.py

VIC-LKH, 2021-12-01:
OK, thank you very much. @tpaviot

closed 2021-12-01 by VIC-LKH