tpaviot/pythonocc-core #1055: Help with pythonocc-core 7.5.1 + ifcopenshell installation

AlonSpinner, 2021-11-16:
Hi Thomas,

I'm Alon, an autonomous robotic's researcher at CEAR lab Technion.
We have had the idea of creating an indoor localization and navigation package based on IFC BIM models, expanding on your work and incorporating it with ROS.

In Short:
I have not been able to install a working environment of pythonocc-core v=7.5.x and ifcopenshell, and am here asking for assistance. 
The problem lies with ifcopenshell.geom.create_shape(settings, product) which I know is not part of your package, but because you clearly have a working environment I decided to turn to you.

Python code and IFC file which fails with my environment can be found here:
https://drive.google.com/drive/folders/13qJx5fx8BMAJ8DmJsRaY5aRZWjC3OZEP?usp=sharing

How I created my setup (16/11/2021):
		in ubuntu 18.04:
		$ conda create --name=pyoccenv751 python=3.8
		$ conda activate pyoccenv751
		$ conda install -c conda-forge pythonocc-core=7.5.1
		$ conda install -c conda-forge -c ifcopenshell ifcopenshell
		$ conda install -c conda-forge pythreejs
		$ conda install -c conda-forge pyqt
		$ conda install -c conda-forge notebook
		$ pip install -U ipykernel #allows debugging in vs code
		$ pip install git+https://github.com/tpaviot/pythonocc-utils

The problem:
	shape = ifcopenshell.geom.create_shape(settings, product) creates a Segmentation fault (core dumped) on everything I have thrown at it.

For a temporary workaround, I have created an environment with pythonocc-core=7.4.0, in which I do not have the segmentation fault, but am not in-sync with current examples found in your demos and elsewhere. This made it quite difficult to extract image data from a renderer.
I'm entering the software world one toe at a time. I hope that with time I can 'speak the language' and also contribute

Thanks,
Alon

tpaviot, 2021-11-17:
Try installing the ifcopenshell I uploaded to my personnal conda channel:
```bash
$ conda install -c tpaviot ifcopenshell 
```

AlonSpinner, 2021-11-18:
Thanks for the quick reply. I fail to install ifcopenshell from your channel due to conflicts.

Here is what I ran:
```
$ conda create --name=pyoccenv751 python=3.8
$ conda activate pyoccenv751
$ conda install -c conda-forge pythonocc-core=7.5.1
$ conda install -c tpaviot ifcopenshell
```
**Output text for the last command:**

Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                          

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versionsThe following specifications were found to be incompatible with your system:

  - feature:/linux-64::__glibc==2.27=0
  - feature:|@/linux-64::__glibc==2.27=0
  - ifcopenshell -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']

Your installed version is: 2.27

**as for libgcc-ng**
using `$ conda list` I have found it is version 9.3.0 that is installed

aothms, 2021-11-20:
What might help is if you pick a ifcopenshell 0.7 which is also built against occt 7.5. There are statically linked archives available here that you can copy into your site-packages folder https://github.com/IfcOpenBot/IfcOpenShell/commit/a89a40e637127404a958983c0687ab90e3e8516a#comments

Or for 0.6 (occt 7.3) https://github.com/IfcOpenBot/IfcOpenShell/commit/721fe4729aa5302efe1602971aae2558934ad098#comments

I have also seen segfaults with certain combinations of pyocc + ifcopenshell. I wish I would understand better, like how these symbols collide and what we can do to prevent it. It's probably an issue similar to: 

> when I go to open an image using the openCV cvLoadImage function, it dies when reading the header, most likely due to differences between headers in 6.2 and 7.0. 
> 
> From: https://stackoverflow.com/questions/2216204/static-and-shared-library-symbol-conflicts

AlonSpinner, 2021-11-28:
Hi aothms, 

Thanks for the build :)
I copied the 3.8 version into my conda-env site-packages and ran into two errors:
1) ' /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found'
2) 'IfcOpenShell not built for 'linux/64bit/python3.8''

From what I read online, ubuntu 18.04 which I am running, does not support GLIBC_2.29. 
I'll try on ubuntu 20.4 and update later this week. The second error might be stemming from the first?

source: https://www.digitalocean.com/community/questions/how-install-glibc-2-29-or-higher-in-ubuntu-18-04

aothms, 2021-11-28:
Oh, right, the 0.6 build uses an older version, so you can use that, or as you say, use 0.7 on ubuntu 20.04.

Correct, the 2nd message just stems from the 1st, failing to load the dynamic library.

---

But I must say, that for a recent project I also had to revert to pyocc 7.4.1, even when I compiled ifcopenshell against the occt from conda. They were memory related issues, perhaps something related to the fact that the conda OCCT depends on TBB, but that IfcOpenShell interfaces with OCCT without TBB...? But I don't know why that would specifically affect pyocc 7.5.1 and not pyocc 7.4.1. @tpaviot any ideas by any chance?

Maozerhouni, 2022-12-30:
> What might help is if you pick a ifcopenshell 0.7 which is also built against occt 7.5. There are statically linked archives available here that you can copy into your site-packages folder [IfcOpenBot/IfcOpenShell@a89a40e#comments](https://github.com/IfcOpenBot/IfcOpenShell/commit/a89a40e637127404a958983c0687ab90e3e8516a#comments)
> 
> Or for 0.6 (occt 7.3) [IfcOpenBot/IfcOpenShell@721fe47#comments](https://github.com/IfcOpenBot/IfcOpenShell/commit/721fe4729aa5302efe1602971aae2558934ad098#comments)
> 
> I have also seen segfaults with certain combinations of pyocc + ifcopenshell. I wish I would understand better, like how these symbols collide and what we can do to prevent it. It's probably an issue similar to:
> 
> > when I go to open an image using the openCV cvLoadImage function, it dies when reading the header, most likely due to differences between headers in 6.2 and 7.0.
> > From: https://stackoverflow.com/questions/2216204/static-and-shared-library-symbol-conflicts

Hi,

I tried your solution but it didn't work for me. Run into ImportError. (I have chosen the folder for py38)

ImportError: IfcOpenShell not built for 'windows\64bit\python3.8'