tpaviot/pythonocc-core #1057: GeomAdaptor_Curve.Curve returns None under Win10

hongyuan1306, 2021-11-19:
Hi,

it seems the function `GeomAdaptor_Curve.Curve` behaves differently under Win10 than under Linux. 

To demonstrate I use the following test file:

```python
from OCC.Core.gp import gp_Pnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
from OCC.Core.BRepAdaptor import BRepAdaptor_Curve

def test():
    p1 = gp_Pnt(5, -5, 0)
    p2 = gp_Pnt(5, 5, 0)
    ed1 = BRepBuilderAPI_MakeEdge(p2, p1).Edge()
    curve = BRepAdaptor_Curve(ed1).Curve()
    print(curve, curve.Curve())

if __name__ == '__main__':
    test()
```

Under Linux `GeomAdaptor_Curve.Curve` returns a  `Geom_Curve` object just as expected:

```
(occ) hong@hong-office:~/workspace/cnc/cnc$ conda list occ
# packages in environment at /home/hong/miniconda3/envs/occ:
#
# Name                    Version                   Build  Channel
occt                      7.5.3                h7391655_0    conda-forge
pythonocc-core            7.5.1            py39h52d5e75_0    conda-forge
(occ) hong@hong-office:~/workspace/cnc/cnc$ python test1.py 
<class 'GeomAdaptor_Curve'> <class 'Geom_Curve'>
```

However under Win10 the same test returns None:

```
(occ) Z:\cnc>
(occ) Z:\cnc>conda list occ
# packages in environment at C:\Users\Administrator\miniconda3\envs\occ:
#
# Name                    Version                   Build  Channel
occt                      7.5.3                h60997fb_0    conda-forge
pythonocc-core            7.5.1            py39h5d9d39a_0    conda-forge

(occ) Z:\cnc>python test1.py
<class 'GeomAdaptor_Curve'> None
```

This seems to be a bug. Is there any other way to get the underlying `Geom_Curve` from a `TopoDS_Edge`?

hongyuan1306, 2021-11-20:
I just ran the test again under matching version between `occt `and `pythonocc-core` and the problem persists:

```
(occ) C:\temp>conda list occ
# packages in environment at C:\Users\hong\miniconda3\envs\occ:
#
# Name                    Version                   Build  Channel
occt                      7.5.1                h60997fb_2    conda-forge
pythonocc-core            7.5.1            py39h5d9d39a_0    conda-forge

(occ) C:\temp>python test.py
<class 'GeomAdaptor_Curve'> None
```

tpaviot, 2021-11-20:
which problem?

hongyuan1306, 2021-11-20:
I found a workaround of this issue is to first save the `BRepAdaptor_Curve` into a python variable and then call `Curve().Curve()` consecutively. The other calling sequences will not work:

```python
from OCC.Core.gp import gp_Pnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge
from OCC.Core.BRepAdaptor import BRepAdaptor_Curve

def test():
    p1 = gp_Pnt(5, -5, 0)
    p2 = gp_Pnt(5, 5, 0)
    ed1 = BRepBuilderAPI_MakeEdge(p2, p1).Edge()
    c1 = BRepAdaptor_Curve(ed1)
    print('This works fine:', c1.Curve().Curve())
    c2 = BRepAdaptor_Curve(ed1).Curve()
    print('But this does not:', c2.Curve())
    print('Neither does this:', BRepAdaptor_Curve(ed1).Curve().Curve())

if __name__ == '__main__':
    test()
```

The results:
```
(occ) C:\temp>python test.py
This works fine: <class 'Geom_Curve'>
But this does not: None
Neither does this: None
```

Strange be seems to be a reliable workaround.

tpaviot, 2021-11-20:
I cannot reproduce the issue on the windows azure builds, et"s close this issue

closed 2021-11-20 by tpaviot