tpaviot/pythonocc-core #1058: Importing issue with Python 3.8 and occ 7.5.1

MrNonCad, 2021-11-19:
Hi All,
I have a issue when using the pyocc-core with python 3.8. 
I cant import GeomLib, i only get the error

**ImportError: DLL load failed while importing _GeomLib: The Specified procedure could not be found**

But I havn't had problems with importing any other module from OCC.Core
And it works with python 3.7

Any idea for a solution to this?

BR
JW

Tanneguydv, 2021-11-19:
Hello, 
Have you specified python 3.8 to your PATH?

MrNonCad, 2021-11-19:
Yes, i have.
Not sure if that has so much meaning since I'm in a conda enviroment when trying to import?
[image]

And as seen above, other modules can be imported and the right python is started.

I have also tried to create an additional enviroment for python 3.9, but GeomLib did not work there neither.

BR

hongyuan1306, 2021-11-20:
With the lastest `pythonocc-core` under Linux python3.9, importing `GeomLib` also failes with the exception:

```
(cnc) hong@hong-matebook:~$ conda list occ
# packages in environment at /home/hong/miniconda3/envs/cnc:
#
# Name                    Version                   Build  Channel
occt                      7.5.3                h7391655_0    conda-forge
pythonocc-core            7.5.1            py39h52d5e75_0    conda-forge

(cnc) hong@hong-matebook:~/workspace/cnc/cnc$ ipython
Python 3.9.7 | packaged by conda-forge | (default, Sep 29 2021, 19:20:46) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.29.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import OCC.Core.GeomLib
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-6f74d90ed7b3> in <module>
----> 1 import OCC.Core.GeomLib

~/miniconda3/envs/cnc/lib/python3.9/site-packages/OCC/Core/GeomLib.py in <module>
     16 # Import the low-level C/C++ module
     17 if __package__ or "." in __name__:
---> 18     from . import _GeomLib
     19 else:
     20     import _GeomLib

ImportError: /home/hong/miniconda3/envs/cnc/lib/python3.9/site-packages/OCC/Core/_GeomLib.so: undefined symbol: _ZN7GeomLib13FuseIntervalsERK18NCollection_Array1IdES3_R20NCollection_SequenceIdEd
```

tpaviot, 2021-11-20:
I cannot reproduce the issue on Linux/py39. The occt version must be 7.5.1, not 7.5.3, please @hongyuan1306 try to install the correct version.

In the latest 7.5.3 occt version, there are very small changes to the API that makes 7.5.1 and 7.5.3 not ABI compliant.

tpaviot, 2021-11-20:
@MrNonCad what's your occt version? Can you please copy/paste the output of the comment ```conda list```.

MrNonCad, 2021-11-20:
Hi,

Ist seems like I also have OCCT 7.5.3 in my enviroment.
[image]

If I create a new enviroment with python 3.7 now, it also get OCCT 7.5.3 and the GeomLib import does not work there neither.
Can I somehow choose OCCT version when I'm installing the package with conda install?

I have installed the package with  "conda install -c conda-forge pythonocc-core=7.5.1"
Using Windows 10 by the way :)

Br
Jw

tpaviot, 2021-11-20:
That's strange, pythonocc-core 751 should come with occt751.

You can try forcing occt to 7.5.1
```conda install -c conda-forge pythonocc-core=7.5.1 occt=7.5.1```

hongyuan1306, 2021-11-20:
After downgrading the `occt` version to 7.5.1, the `GeomLib` import works fine.

tpaviot, 2021-11-20:
I confirm the issue on linux py39. The occt version installed is 7.5.3 and has to be forced to 7.5.1. I will change the README accordingly.

MrNonCad, 2021-11-20:
I did a conda install for only occt=7.5.1 in my already exeisting enviroment and now it works fine.

Big thanks for your help!

Br
Jw

closed 2021-11-20 by tpaviot