tpaviot/pythonocc-core #1063: Versions & wrapper

ageeye, 2021-11-28:
For me is only the build with occ@7.5.2 possible. So, I have some questions:

1. In the interests of clarity, is it possile to tag the commit like 'Port to occt-7.5.3.' in the future?
2. Is the wrapper or the python version cause of this trouble? I use python python@3.9.7.

occ 7.5.3
=========

```
/tmp/pythonocc-core-20211128-42284-1sx3642/Build/Darwin/arm64-Release-/MMgtPYTHON_wrap.cxx
/tmp/pythonocc-core-20211128-42284-1sx3642/Build/Darwin/arm64-Release-/FSDPYTHON_wrap.cxx:3362:10: fatal error: 'Standard_Failure.hxx' file not found
#include <Standard_Failure.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/_FSD.dir/Darwin/arm64-Release-/FSDPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/_FSD.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/tmp/pythonocc-core-20211128-42284-1sx3642/Build/Darwin/arm64-Release-/MMgtPYTHON_wrap.cxx:3361:10: fatal error: 'Standard_Failure.hxx' file not found
#include <Standard_Failure.hxx>
```

occ 7.6.0
=========

```
tmp/pythonocc-core-20211128-47358-djfnvu/Build/Darwin/arm64-Release-/MMgtPYTHON_wrap.cxx
In file included from /tmp/pythonocc-core-20211128-47358-djfnvu/Build/Darwin/arm64-Release-/FSDPYTHON_wrap.cxx:3531:
/tmp/pythonocc-core-20211128-47358-djfnvu/src/SWIG_files/headers/FSD_module.hxx:23:9: fatal error: 'FSD_Base64Decoder.hxx' file not found
#include<FSD_Base64Decoder.hxx>
        ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /tmp/pythonocc-core-20211128-47358-djfnvu/Build/Darwin/arm64-Release-/MMgtPYTHON_wrap.cxx:3530:
/tmp/pythonocc-core-20211128-47358-djfnvu/src/SWIG_files/headers/MMgt_module.hxx:22:9: fatal error: 'MMgt_TShared.hxx' file not found
#include<MMgt_TShared.hxx>
        ^~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/_FSD.dir/Darwin/arm64-Release-/FSDPYTHON_wrap.cxx.o] Error 1
```
   
  Thanks for your support.

tpaviot, 2021-12-01:
@ageeye The 7.5.3 tag will be added when the library is ready to be released. There are still quite a few required fixes before that, but I hope it can be done in a very short term (by the end of this year)

The current master branch has to be built on occt-7.5.3

ageeye, 2021-12-01:
Thank you for your explanation.