tpaviot/pythonocc-core #1064: Setting

AlonSpinner, 2021-11-28:
Hi again,

I'm trying to calibrate the offline renderer's camera with my own intrinsic parameters (focal length and others)
Noticing that the projection matrix is in OpenGL format, I found a way to do it with this:
https://strawlab.org/2011/11/05/augmented-reality-with-OpenGL/

Unfortunately, I found no set method for the camera's projection matrix.
Any idea how I can go about it?

Thanks,
Alon

Code to extract projection matrix from a camera object:
```
from OCC.Display.OCCViewer import Viewer3d
from ctypes import c_double
import numpy as np
offscreen_renderer = Viewer3d()
offscreen_renderer.Create()
offscreen_renderer.SetModeShaded()

cam = offscreen_renderer.View.Camera()
data = cam.ProjectionMatrix().GetData()
v = list((c_double * 16).from_address(int(data)))
M = np.array(v).reshape((4,4))
print(M)
```

tpaviot, 2021-12-01:
You can use the ```.SetValue``` method

tpaviot, 2021-12-01:
According to the official documentation https://dev.opencascade.org/doc/refman/html/class_graphic3d___camera.html I can't see any method to directly set the projection matrix

tpaviot, 2021-12-03:
After a short review, I unfortunately did not find any solution

AlonSpinner, 2021-12-03:
I'll try to make a workaround function SetCameraIntrinsics(fx,fy,px,py) with these:
```
cam.SetDistance
cam.SetFOVy
cam.SetZFocus
cam.SetZRange
cam.SetIOD
cam.SetScale
cam.SetTile
cam.SetAspect
```
I found that the projection is similar to openGL, but not quite.
I get this matrix from occ:
```
[[ 0.05363987,  0.        ,  0.        ,  0.        ],
 [ 0.        ,  0.07151983,  0.        ,  0.        ],
 [ 0.        ,  0.        , -0.06425702,  0.        ],
 [-0.        , -0.        , -1.99999684,  1.        ]]
```

This is how it is defined in openGL
http://www.songho.ca/opengl/gl_projectionmatrix.html

AlonSpinner, 2021-12-03:
Using : http://www.songho.ca/opengl/gl_projectionmatrix.html

def setOrthographicProjectionMatrix(cam,scale,aspect,n,f):
    # P1[0,0] = 2/scale
    # P1[1,1] = aspect*2/scale
    # P1[2,2] = -2/(f-n)
    # P1[2,3] = -(f+n)/(f-n)
    cam.SetProjectionType(cam.Projection_Orthographic)
    cam.SetZRange(n,f) #affects P1[2,2] P1[3,2]
    cam.SetScale(scale) #affects P1[1,1] P1[0,0]
    cam.SetAspect(aspect) #affects P1[1,1] P1[0,0]