tpaviot/pythonocc-core #1065: class TColStd_PackedMapOfInteger not wrapped

cedricadam2, 2021-12-02:
Cannot use the class TColStd_PackedMapOfInteger.
I obtain this object: <Swig Object of type 'TColStd_PackedMapOfInteger *' at 0x000002298EBDAD50>

tpaviot, 2021-12-02:
I cannot reproduce the issue with current master branch:
```bash
>>> TColStd_PackedMapOfInteger
<class 'OCC.Core.TColStd.TColStd_PackedMapOfInteger'>
>>> TColStd_PackedMapOfInteger()
<class 'TColStd_PackedMapOfInteger'>
>>> 

```

cedricadam2, 2021-12-02:
I am using pythonocc-core 7.5.1 on win10.
I cannot import the class TColStd_PackedMapOfInteger, I obtain this error:
```
>>> from OCC.Core.TColStd import TColStd_PackedMapOfInteger
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'TColStd_PackedMapOfInteger' from 'OCC.Core.TColStd' (C:\Users\cedri\Documents\Projet_Geoptim\geoptim\geoptim_env\lib\site-packages\OCC\Core\TColStd.py)
```
but I can import the class TColStd_HPackedMapOfInteger:
```
>>> from OCC.Core.TColStd import TColStd_HPackedMapOfInteger
>>> TColStd_HPackedMapOfInteger
<class 'OCC.Core.TColStd.TColStd_HPackedMapOfInteger'>
```
So it seems to be fixed in current master branch.

cedricadam2, 2021-12-02:
Added in commit 6b160b5

tpaviot, 2021-12-02:
oh yes, I remember

closed 2021-12-03 by tpaviot