tpaviot/pythonocc-core #1068: STEPCAFControl_Reader().Transfer() taking too long to execute

Sadiq-Naizam, 2021-12-09:
So Im trying to convert a STEP file to XDE Document. The file size is around 115MB. It takes around 90 seconds to execute the STEPCAFControl_Reader().transfer() function. But it doesnt take this long when I tried the same using FreeCAD or other open source CAD softwares that uses OCC. 

My device spec is :
RAM : 16GB
Processor: Intel(R) Core(TM) i7-10875H @ 2.30GHZ (16 CPUs)
NVIDIA GeForce RTX 2080 Super
VRAM : 8GB

Also i noticed that python isnt taking advantage of my CPU. Is this because of the Global Interpreter Lock? the transfer() function is implemented in C++ right, so it should be affected by GIL right?

zeroright, 2021-12-09:
I also meet this same problem.

tpaviot, 2021-12-15:
How long does it take on FreeCAD compared to pythonocc? They both use the same API, there should not be any big difference in terms of performance. Which version of pythonocc do you use?

Sadiq-Naizam, 2021-12-15:
@tpaviot The model loads up under 30 - 45 sec on FreeCAD. I'm using version 7.5.1 of pythonocc.
But it takes between 75-90 seconds to convert from STEP to XDE. Is this the expected time for converting a 115MB file? 
Thanks

roosre, 2022-06-29:
Same here. I just tested OCC 7.6.2 (c++) and there is no improvement if compared with 7.1. I thought the issue is addressed
https://tracker.dev.opencascade.org/view.php?id=29830

dpasukhi, 2024-09-25:
The STEP import was improved up to 75% on the big model with OCCT 7.8.
We plan to have more improvements for OCCT 7.9.