tpaviot/pythonocc-core #1075: Build with occt 7.6.0

Krasner, 2021-12-26:
occt 7.5.3 is not compatible with my conda environment, but occt 7.6.0 is.
When I try to build pythonocc with occt 7.6.0 I get the following error:

`pythonocc-core/src/SWIG_files/headers/TCollection_module.hxx:24:9: fatal error: TCollection_BaseSequence.hxx: No such file or directory
 #include<TCollection_BaseSequence.hxx>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/_XmlTObjDrivers.dir/build.make:76: CMakeFiles/_XmlTObjDrivers.dir/Unix/x86_64-Release-/XmlTObjDriversPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1069: CMakeFiles/_XmlTObjDrivers.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`

I believe opencascade 7.6.0 no longer has that header file.

Is there any plan to make a build that is compatible with occt 7.6.0? 

Thanks

Krasner, 2021-12-26:
This is not the only header file that is missing in opencascade 7.6.0

sfornengo, 2022-01-21:
Any progress ?

Krasner, 2022-01-23:
There has not been an update on pythonOCC. trimesh and gmsh are built against the latest opencascade version. Actually trimesh uses gmsh under the hood.

closed 2022-01-23 by Krasner

tpaviot, 2022-08-18:
Updated in current master branch, current release support occt7.6.2