tpaviot/pythonocc-core #1076: Determine whether a point lies within a shape

hongyuan1306, 2021-12-27:
Hi,

I have a closed wire on the `XOY` plane loaded from `STEP` file and I am trying to determine whether a given point on that plane is inside the wire or not using `IntTools_FClass2d`. I am not sure whether this is the correct tool to use since I find that under some circumstances a point lying outside of the bounding box of the shape is classified as an inside point, and I do not understand why.

The test files can be found in the attached zip file [test.zip](https://github.com/tpaviot/pythonocc-core/files/7779629/test.zip). Following is the code:

```python
from OCC.Core.STEPControl import STEPControl_Reader
from OCC.Core.TopoDS import TopoDS_Shape
from OCC.Core.Bnd import Bnd_Box
from OCC.Core.gp import gp_Pnt2d
from OCC.Core.GeomAbs import GeomAbs_Arc
from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakeOffset
from OCC.Core.BRepMesh import BRepMesh_IncrementalMesh
from OCC.Core.BRepBndLib import brepbndlib_Add
from OCC.Extend.TopologyUtils import TopologyExplorer
from OCC.Core.IntTools import IntTools_FClass2d
from OCC.Core.TopAbs import TopAbs_IN
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeFace

def show_bounding_box(shape: TopoDS_Shape):
    mesh = BRepMesh_IncrementalMesh()
    mesh.SetParallelDefault(True)
    mesh.SetShape(shape)
    mesh.Perform()
    if not mesh.IsDone():
        raise AssertionError('Mesh not done.')
    bbox = Bnd_Box()
    bbox.SetGap(1e-10)
    brepbndlib_Add(shape, bbox)
    print('Bounding box', bbox.Get())

def main():
    step_reader = STEPControl_Reader()
    step_reader.ReadFile('test1.step')
    step_reader.TransferRoot()
    shape = step_reader.Shape()

    explorer = TopologyExplorer(shape)
    wire = next(explorer.wires())

    maker = BRepOffsetAPI_MakeOffset()
    maker.Init(GeomAbs_Arc)
    maker.AddWire(wire)
    maker.Perform(-10)
    shape = maker.Shape()
    show_bounding_box(shape)

    maker = BRepBuilderAPI_MakeFace(shape)
    face = maker.Face()
    classifier = IntTools_FClass2d(face, 1e-10)

    state = classifier.Perform(gp_Pnt2d(0, 92), True)
    print(
        f'Point (0, 92) is {"inside" if state==TopAbs_IN else "outside"} the shape'
    )

if __name__ == '__main__':
    main()
```

The test result is shown below:

```
(cnc) hong@hong-office:~/workspace/cnc/cnc$ python test1.py 
Bounding box (-176.2819439651443, -110.0010001, -0.0010001, 176.28194396516432, 88.00100010000007, 0.0010001)
Point (0, 92) is inside the shape
```

According to the test, the point (0, 92) is classified to be inside the shape, but the shape is bounded by a maximal y coordinate of 88. Why this happens and how to fix it?

Your help is appreciated.

cafhach, 2022-01-26:
You have to distinguish the UV parametric space of the surface to the XYZ euclidian space in which the surface is embedded. The classifier `IntTools_FClass2d` considers only the UV parametric space. The coordinate (0, 92) in UV coordinates can end up anywhere in your XYZ space, depending on the definition of the surface.

If you want to classify a 3d point with respect to a body, you can use `BRepClass3d_SolidClassifier`. But note that this requires your CAD model to be described as a `TopoDS_Solid`. Some CAD software exports bodies as a loose collection of faces. Hope this helps.

hongyuan1306, 2022-02-17:
@cafhach  

Thanks for pointing out the difference between UV space and XYZ space. Now I understand I have to project the XYZ point onto the face to find out UV parameters, and then use the UV parameters as input to `IntTools_FClass2d`.

The classification is now working as expected.

closed 2022-02-17 by hongyuan1306