tpaviot/pythonocc-core #1077: `angular_deflection` in `write_stl_file` doesn't work when Use the command twice at once with different values

QHQIII, 2022-01-03:
I am trying to control the file size by `angular_deflection` in `write_stl_file`, but when I Use the command twice at once with different values of `angular_deflection`， I can only get two files of the same size. Did I make a mistake?
```
i="C:\\Users\\BREEZ\\Desktop\\emp\\3D\\3D15-H06.step"
o1="C:\\Users\\BREEZ\\Desktop\\temp\\3D\\3.stl"
o2="C:\\Users\\BREEZ\\Desktop\\temp\\3D\\4.stl"
shapes = read_step_file(input_filepath)
write_stl_file(shapes,o1,"binary",0.1,0.5)
write_stl_file(shapes,o2,"binary",0.5,0.5)
```
[image]

cafhach, 2022-01-26:
Just a theory: If the precision of your STL file was fully determined by the linear deflection (because it was so small) then `angular_deflection` could have no effect. You could check this theory by increasing `linear_deflection` considerably.

gkv311, 2022-01-26:
You may try calling `BRepTools::Clean()` on a shape before each `write_stl_file()` to ensure old triangulation is not reused (don't know what `write_stl_file()` does internally).