tpaviot/pythonocc-core #1078: Cannot create "MeshVS_EntityType &"

cedricadam2, 2022-01-09:
I want to have access to mesh geometry using :
` Coords = TColStd_Array1OfReal()
  Type = MeshVS_EntityType.MeshVS_ET_Node
  NbNodes = data_source.GetGeom (ID, is_element, Coords, Type)`
but I obtain this error :
` NbNodes = data_source.GetGeom (ID, is_element, Coords, [MeshVS_EntityType.MeshVS_ET_Node])
  File "C:\Users\cedri\Documents\Projet_Geoptim\geoptim\geoptim_env\lib\site-packages\OCC\Core\MeshVS.py", line 2145, in GetGeom
    return _MeshVS.MeshVS_DataSource_GetGeom(self, *args)
TypeError: in method 'MeshVS_DataSource_GetGeom', argument 6 of type 'MeshVS_EntityType &'`

How can I create a "MeshVS_EntityType &" variable ?

tpaviot, 2022-01-11:
In the current master branch, the ```GetGeom``` method takes only three arguments. The type is returned.

cedricadam2, 2022-01-11:
Thank you, it works on current master branch.