tpaviot/pythonocc-core #1088: Is OpenCascade 7.5.2 still required for pythonOCC compilation? Or can I use a newer version like 7.6.0?

visusys, 2022-03-01:
Outlined here:

[https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md](https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md)

> **Requirements**
> pythonOCC needs the following libraries or programs to be installed before you can compile/use it :
> The python programming language (http://www.python.org/)
> OpenCascade 7.5.2 (https://dev.opencascade.org/)
> (snip)

Question is the title of this issue. Do we have a green light to use OpenCascade 7.6.0, or does pythonOCC render the older version necessary?

Thanks for any assistance.

rainman110, 2022-03-02:
As pythonOCC is a direct mapping of the opencascade C++ API, it is directly tied to a specific API version, namely 7.5.2.

To update to OCCT 7.6.0, the swig bindings might need to be adapted / recreated.

Krande, 2022-04-22:
Hey, for what its worth I am experimenting with adapting the SWIG bindings for OCCT 7.6.1. You can follow the progress in https://github.com/tpaviot/pythonocc-generator/issues/84

closed 2022-08-23 by tpaviot