tpaviot/pythonocc-core #1089: STEPControl_Reader lead to shape deformation

linbeijianbaoxia, 2022-03-03:
As mentioned in the topic，I use the demo to load my STEP file, but the shape of holes has changed. But, when I open it in FreeCAD, it go well. As we know, FreeCAD also use OCCT.
[image]
[image]
 @tpaviot

linbeijianbaoxia, 2022-03-07:
today, I try to fix my model with the OCC.Core.ShapeFix package，and happy to find that it works in some degree. However, there still exist few problems， such as the holes are not cylinders. May be there are some other ways. There are my fix demo and picture。

`from OCC.Core.ShapeFix import ShapeFix_Shape

thePrec = 1e-7
theMaxTol = 1
theMintol = 1

aFixShape = ShapeFix_Shape()
aFixShape.Init(blade_shape)
aFixShape.SetPrecision(thePrec)
aFixShape.SetMaxTolerance(theMaxTol)
aFixShape.SetMinTolerance(theMintol)

aFixShape.Perform()
aResult = aFixShape.Shape()`

[image]
[image]

dpasukhi, 2024-09-25:
Hello, I guess there are not issue with healing.
The issue with mesh generation. for some reason deflection is too high