tpaviot/pythonocc-core #1094: Spring shape generation failed

luoyu-123, 2022-03-28:
I want to generate a shape of a spring, but when I scan along the thread using the BRepOffsetAPI_MakePipe function the model is incomplete and I don't know what's wrong. The version I am using is 7.4.0
The picture is as follows：

[image]

cafhach, 2022-04-22:
Looks like the spring is fully created but it only visualizes every second segment as a mesh and the others using a simplified style. Can you show your code to visualize the object?

luoyu-123, 2022-04-23:
Thank you very much for your help, this and the problem have been solved on my side. It is a tolerance parameter mode that is set for less entities generated by scanning

[image]

best wishes！

closed 2022-04-23 by luoyu-123